NVIDIA-AI-IOT/jetbot #212: Different type of Jetpack

sreba, 2020-05-26:
What is "JetPack " that I foind in the first step in software setup senction-
"Step 1 - Flash JetBot image onto SD card
Download the expandable JetBot SD card image jetbot_image_v0p4p0.zip

The above image is based on JetPack 4.3. The previous image based on JetPack 4.2 may be found here: jetbot_image_v0p3p2.zip."

tomMEM, 2020-06-19:
Hello, JetPack contains the operation system to run for example the Jetson nano. It also contains libraries such as CUDA, TensorRT etc.. It is updated from time to time, which gives the different versions. 
On a specific JetPack, the developer can add their specific Python, Matlab libraries. However, because of different versions of the individual libraries, it can create a huge amount of trial and error and frustration to get the right set of libraries and scripts installed.

Thus, the best is to use just the provided images from JetBot, Donkey Racercar or Jetsonhack that contain the required operation system and libraries and run smoothly with e.g. JetPack 4.3. 

For more information :https://docs.nvidia.com/jetson/jetpack/install-jetpack/index.html
https://developer.nvidia.com/jetpack-43-archive

thIYan-EsWar, 2021-11-15:
What are the differences between JetPack OS and the JetBot OS?