NVIDIA-AI-IOT/jetbot #218: How to build a virtual JetBot environment on PC?

tomMEM, 2020-06-19:
Hello, what would be a clean approach to install e.g. the JetBot library and script set in a virtual environment on a PC?  
Is is possible to run the SD image in a Dockerfile with all the dependencies? I thought to do the training on a power PC but it seems difficult to get all the right libraries and scripts.
Would be great to get a hind for a direction how to do.
T

tedchina, 2020-06-29:
IMHO, this can hardly work.
Nano has an Arm CPU and a Nvidia GPU, the training depends on CUDA a lot. You have to set up your PC differently from Nan's configuration even if your PC is with a Nvidia GPU.

On the other side, if you just want to read and even edit the Jupyter Notebook on your PC, simply copy the Notebook and jetbot folders are enough.
You may need to have install Python3 and Jupyter Lab on your PC if not available yet.

tomMEM, 2020-07-02:
Thank you for your note and response pointing me to check if one of my PCs has actually a Cuda 10 compatible graphic card - unfortunately only my notebook has. I found https://developer.download.nvidia.cn/compute/cuda/10.0/Prod/docs/sidebar/CUDA_Installation_Guide_Windows.pdf that explains the installation process for Cuda and dependencies, but it seems others recommend to use a virtual Ubuntu and install there. Still will give it a trial to learn how to run the jetbot "training script" on a different OS.
T

jaybdub, 2020-07-10:
Hi tomMEM,

Thanks for reaching out.

The training script itself is fairly light on dependencies.  Primarily, it requires PyTorch.  You can try installing this by following the desktop instructions

https://pytorch.org/get-started/locally/

Or by using a docker container packaged with the requirements 

https://ngc.nvidia.com/catalog/containers/nvidia:pytorch

The NGC pytorch docker container includes Jupyter Lab, which is needed for the training notebook.

Please let me know if this helps, or you run into any issues along the way. Happy to help how I can.

Best,
John

tomMEM, 2020-07-14:
Hello John, thank you for your time and pointing into the direction. I tried VM VirtualBox  from Oracle on Windows with a docked pytorch container. Seems to be a good start to learn more without major installations.
Best
T

closed 2022-07-15 by jaybdub