NVIDIA-AI-IOT/jetbot #220: How to stop automatic adjustment of camera's sensitivity or exposure time?

tomMEM, 2020-06-20:
Hello, I wonder how to fix the sensitivity or exposure time of a camera connected to the nano B01.
The testing environment has different light levels and the objects are rather dark causing the camera to adjust sensitivity or exposure time. The adjustment takes a couple of frames and delays response time or causes back-forward for a second within the different tasks.
Any suggestions how to fix the camera settings?
Thanks. T

jaybdub, 2020-07-10:
Hi tomMEM, 

Thanks for reaching out!

The camera is wrapping a GStreamer pipeline, which uses the ``nvarguscamerasrc`` GStreamer element to capture images.  You can see the GStreamer pipeline string on this line of code

https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/camera.py#L50

If the JetBot Camera interface doesn't expose what you need for your purposes, you may be able to modify this pipeline directly (by modifying that line of code).

To see which parameters can be modified, try typing

```bash
gst-inspect-1.0 nvarguscamerasrc
```
From a terminal.  It includes the exposure time range.

For example, you could try modifying the code like this

```python
    def _gst_str(self):
        return 'nvarguscamerasrc exposuretimerange="34000 358733000"  ! video/x-raw(memory:NVMM), width=%d, height=%d, format=(string)NV12, framerate=(fraction)%d/1 ! nvvidconv ! video/x-raw, width=(int)%d, height=(int)%d, format=(string)BGRx ! videoconvert ! appsink' % (
                self.capture_width, self.capture_height, self.fps, self.width, self.height)
```

Please note, I haven't tested this.  I'm interested to hear your results.

You can find more about how to construct GStreamer pipelines here

https://developer.download.nvidia.com/embedded/L4T/r32_Release_v1.0/Docs/Accelerated_GStreamer_User_Guide.pdf?8dvdvNdtpiftrL02TEUQ4g_X25QvM4Ib-zREdRaueLLJNPth1DIjEKHUaBi0qA3Jkh_26K032iTEq__uzYTAKZ5DgPqgj9tpUMH7RbKKn-PK1NBC64EgLjiKwCBfVIveg5YNshT5aY0O_8Rogjet1D83abPitY5vDWDWxNMuQKeTZ4BE-7Q

Please let me know if this helps or you run into any issues.  I'm happy to provide more details if needed

Best,
John

tomMEM, 2020-07-12:
Hello John, thank you for your time and notes.
I tried unprofessionally different settings to prevent Auto adjustment functions for gain, white balance and exposure and other settings.
At least for the collision example the detection of blocking objects (dark) did not lead to Auto adjustment-related false object recognition, so it improved. I had to re-train with the new settings a bit.
The only inconvenience now is to adjust manually the exposure or gain range, but because I test at the same settings it does not  matter much.
At the end "camer.unlink" gives an additional boost - the system seems now to detect within few milliseconds. 

return 'nvarguscamerasrc sensor-mode=4 awblock=false maxperf=true wbmode=1 exposuretimerange="37000000 37000000" aeantibanding=0 ispdigitalgainrange="8 8" gainrange="8 8" ! video/x-raw(memory:NVMM), width=%d, height=%d, format=(string)NV12, framerate=(fraction)%d/1 ! nvvidconv ! video/x-raw, width=(int)%d, height=(int)%d, format=(string)BGRx ! videoconvert ! appsink' % (

jaybdub, 2020-07-13:
Hi tomMEM,

Glad to hear you got some performance improvements.

Was the ``camera.unlink`` to remove the camera streaming to the Image widget, or something else?

Best,
John

tomMEM, 2020-07-14:
Hello John, thank your for your time.
Preventing the live stream video window by either disabling display(image) directly or camera.unlink later improved performance speed. I do not have a benchmark measure, so just visual - might be it could be added a stop watch for free-blocking event time difference to get a number.

Besides, the call display(image) seems to slow down the system and also leads to memory build up. I tried several browsers, but they seem to use all a similar rendering engine - so not difference. Once I tried the Epiphany browser that has a rendering engine PID and nicely shows the build up.  I am not sure if a matplot display call would create a similar behavior with a live video stream in a browser window.
Best
T

closed 2022-07-15 by jaybdub