NVIDIA-AI-IOT/jetbot #226: Motors not working

Kieran-42, 2020-07-16:
Hello. We are using a preassembled/presoldered FeatherWing suggested in the bill of materials. We have downloaded the proper software and used only the hardware recommended. We see the featherwing at the address 0X60 and 0X70 using the i2c detect command.  While running the basic operation and teleoperation notebooks we have seen no error messages. Despite that whenever we run the code the motors do not respond. Any pointers or suggestions would be most welcome thank you!

culebracut, 2020-07-18:
Hi, I am having the same issue with the motors not responding.  What is syntax of the i2c detect command?  I can try the same.  The camera, wifi, lcd display all work fine.  I have a multi-meter and never see voltage.

culebracut, 2020-07-19:
For me the fix was to use the Waveshare software instead of generic Nvidia.  

So I changed  this:  git clone https://github.com/NVIDIA-AI-IOT/jetbot
to this: git clone https://github.com/waveshare/jetbot

greenham592, 2020-11-06:
Hi, I am stuck here as well. I am using a Nano 2GB and a Waveshare kit. The camera, LCD work fine. I have used both the Waveshare and Nvidia Jetbot clones. I did check the motors with a battery, and they do work. No error messages are shown. Any help would be appreciated.

closed 2022-07-15 by jaybdub