NVIDIA-AI-IOT/jetbot #227: Link motors to traitlets - Jetbot

ai-hengineer, 2020-07-17:
I’m trying out Jetbot example 1 (basis_motions.ipynb) on my Sparkfun Jetbot.
The motors are working well when set «directly» through methods: left, right, forward, backward.
However, when linking the motors to widgets using traitlets (using the example), the motor does not move.
Does anyone have a solution for this?

santaimpersonator, 2020-08-05:
For the SparkFun Jetbot:

With the traitlets, the motors should be controlled by the [motor.py](https://github.com/sparkfun/jetbot/blob/master/jetbot/motor.py) file. I just tested a fresh v01-10 image on a Jetson Nano (with the one camera connector carrier board), but didn't have issues.

1. Did you link the traitlets multiple times? I can't remember, but I think this was a little finicky and you had to link it either once or twice for the traitlets to work.
2. Let's try to double check the motor driver is recognized. Enter the following in a terminal: `i2cdetect -y -r 1`. You should see an address listed at `5D`. If not, you may need to double check that the Qwiic cable.
3. Let's check how the motor driver is configured (you if you could check this at various points; it would help to find where the code might be breaking). Enter the following in a terminal: `i2cdump -y -r 1 0x5D`.
    * At register `SCMD_STATUS_1 = 0x77` you will see:
        * `0`: When the motor driver is ready.
        * Greater than `0`: When the motor driver is enumerating or busy.
    * At register `SCMD_DRIVER_ENABLE = 0x70` you will see:
        * `0`: When the motor driver is disabled.
        * `1`: When the motor driver is enabled.
    * At register ` SCMD_MOTOR_A_INVERT = 0x12` and `SCMD_MOTOR_B_INVERT = 0x13` you will see:
        * `0`: When the motor driver direction is not inverted.
        * `1`: When the motor driver direction is inverted.
    * At register SCMD_MA_DRIVE = 0x20` and `SCMD_MB_DRIVE = 0x21` you will see:
        * `0x80`: When the motor driver speed is set to `0`.
        * <`0x80`<: When the motor driver direction is not `0`.
4. Let's see if there might be something weird happening in the code between the two cells:
    1. Try restarting and clearing the kernel.
    2. Then, implement the `from jetbot import Robot` cell.
    3. Skip the `robot.left(.3)` `time.sleep(.5)` `robot.stop()` cell.
    4. Proceed on to the widget and traitlet linking cells.

santaimpersonator, 2020-10-09:
@ai-hengineer If this issue has been resolved please close this issue.

closed 2022-07-15 by jaybdub