NVIDIA-AI-IOT/jetbot #228: road following cannot very good implementation

hope450, 2020-07-20:
This road following demo cannot achieve the expected results . I don’t know if it is caused by insufficient data collection or insufficient accuracy. Can I give some suggestions for improvement?

jaybdub, 2020-07-21:
Hi hope450,

Thanks for reaching out!

Sure, I'm happy to help how I can.  Could you share details on what you have tried so far?  I may be able to provide more detailed guidance then.

That said, some general tips:

1.  Data variance is _essential_.  This is one of the biggest lessons we hope to teach with JetBot.  Sometimes a model trained with 5000 similar images will be much worse than a model trained with 100 well varied images.
    1.  Different angles
    2.  Different lighting conditions
2.  Make sure to let training converge
    1.  Stopping the model too early may be the problem
3.  Test before running
    1.  Sometimes it's hard to diagnose the issue when running the full live demo.  Try turning off the motors, and just placing the robot in different spots, at different angles to see how the neural network responds.  If this doesn't work well, I wouldn't expect the full demo will.

Please let me know if this helps or you have any further questions.

Best,
John

BobbyLCW, 2020-07-31:
Hi @jaybdub ,

I have 1 question need to ask regard the code inside jupyter lab (road following training).
[image]
from the description above mention that input is 512 and output is 1 (regression). 
So, why does the model.fc = torch.nn.Linear(512, 2) ? the output feature 2 is valid? or i should change to 1 instead?
I am totally new to pytorch anyway. Thanks.

abuelgasimsaadeldin, 2020-08-09:
Hi John,

Just wanted to ask on your third point you mentioned that it will be best to test how your neural network responds at different angles while turning off the motors. Just wanted to know how exactly do you do that?

abuelgasimsaadeldin, 2020-08-18:
Hi all,

I am still facing the problem of my road following live demo not achieving the expected results and I am still wondering how I can identify the problem by testing my model and seeing how it reacts as I place the Jetbot in different places around the circuit. Any help will be much appreciated.

fqez, 2020-09-09:
Hi @abuelgasimsaadeldin 

I think what @jaybdub is saying is that you can comment out this two lines:

```
robot.left_motor.value = max(min(speed_slider.value + steering_slider.value, 1.0), 0.0)
robot.right_motor.value = max(min(speed_slider.value - steering_slider.value, 1.0), 0.0)
```
this way the robot will not move when you run the live demo. Then you can see the values given by the model in the steering and speed sliders. Place the robot on a straight segment of your circuit and check if the steering value is low (about 0) and the speed value is high (about 1); then place the robot on a curve and check if the steering value on the slider is high and the speed is low, and so on.
If the model gives you good predictions, chances are that the example will work as expected, otherwise try checking the 3 points that @jaybdub put in its first comment.

Hope it helps!

closed 2022-07-15 by jaybdub