NVIDIA-AI-IOT/jetbot #229: Power Management

RohanPhadnis, 2020-07-23:
Hello,

I am using the recommended power bank and an ir camera from the CSI port. Besides these 2 connections, I also connected to my monitor using HDMI. The Jetson is set to 5W mode. However, despite using such few peripherals, the Jetson Nano is not starting. It keeps shutting down due to a lack of enough power. Help!

Thanks in advance,
Rohan

VegardIversen, 2020-07-23:
Hi @RohanPhadnis 

Im guessing you have flashed an sd card with an image (it wont start unless it has a image).
What kind of microusb cable are you using? From my experience you have to use a high quality cable (and a short one).

RohanPhadnis, 2020-07-23:
Thank you so much for the response! Yes, I have flashed the image to the card. I will try using a better and shorter cable though.

VegardIversen, 2020-07-23:
And remember to check the output of the power bank. Some have one output with high voltage and ampere, and the other with low.  

Also useful to have a raspberry pi power supply to debug with (From my experience, this gives sufficient voltage and ampere). 
Might be some useful information on this link: [https://www.raspberrypi.org/forums/viewtopic.php?t=165393](url)

RohanPhadnis, 2020-07-23:
@VegardIversen, your tip was amazing! I changed the cable and it worked! Thank you so, so, so, so much!!!!!

closed 2020-07-23 by RohanPhadnis