NVIDIA-AI-IOT/jetbot #231: AttributeError: 'Motor' object has no attribute '_motor'

VegardIversen, 2020-07-23:
Hi
Im suddenly getting this error when running robot.stop():
 ```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/jetbot-0.3.0-py3.6.egg/jetbot/motor.py", line 60, in _release
    self._motor.disable()
AttributeError: 'Motor' object has no attribute '_motor'
``` 
Any suggestions on how to fix?

jaybdub, 2020-07-23:
Hi VegardIversen,

Thanks for reaching out!

Are you using one of the JetBot SD card images, or perhaps a third party JetBot with a different set of instructions?

If you're using a third party bot, do you mind sharing the instructions you followed?

Best,
John

VegardIversen, 2020-07-23:
> Hi VegardIversen,
> 
> Thanks for reaching out!
> 
> Are you using one of the JetBot SD card images, or perhaps a third party JetBot with a different set of instructions?
> 
> If you're using a third party bot, do you mind sharing the instructions you followed?
> 
> Best,
> John

Hi
Thank you for quick replay. Im using the image from this kit [https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit/all](url). T
The weird thing is that it worked before.

jaybdub, 2020-07-23:
Hmm, strange.

Did you update or modify the software at all?

If not, I would suspect it is a hardware issue.  Perhaps the motor driver connection came loose, and is not discoverable.

You can use the i2cdetect tool to see if the motor driver is discoverable.

Do you mind sharing the output of the following

```bash
sudo i2cdetect -y 1
```

Best,
John

VegardIversen, 2020-07-24:
> Hmm, strange.
> 
> Did you update or modify the software at all?
> 
> If not, I would suspect it is a hardware issue. Perhaps the motor driver connection came loose, and is not discoverable.
> 
> You can use the i2cdetect tool to see if the motor driver is discoverable.
> 
> Do you mind sharing the output of the following terminal command
> 
> ```shell
> sudo i2cdetect -y 1
> ```
> 
> Best,
> John

I have just updatet opencv. Motors seems to work driving forward, but gets an error when using robot.stop() and they dont drive the correct duration. 

This is the output from the command: 
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                                                 
20:                                                 
30: -- -- -- -- -- -- -- --                         
40:                                                 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60:                                                 
70:

santaimpersonator, 2020-08-05:
Hey there, you seem to be having an issue similar to another user on the [SparkFun forum](https://forum.sparkfun.com/viewtopic.php?f=88&t=53479).

I just tested this out on our newer v01-10 image and with a Jeston Nano (with the one camera connector carrier board), but I wasn't able to duplicate the issue. Did you happen to modify the files at all? Here is a link to our [fork of this repo](https://github.com/sparkfun/jetbot) for a reference.

If you accidentally modified the files, you can follow these instructions for building a card from scratch: https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch
* Start from step 9, but use the link to our repo instead of cloning Nvidia's.
* If you clone Nvidia's repo, you will run into issues because of the hardware difference.

VegardIversen, 2020-08-06:
Hey, that is also my post. The error are just showing up after a program has run, and not during so its not breaking the system anymore. I will try with new hardware on another unit soon, and can see if that fixes the problem.

santaimpersonator, 2020-08-06:
As you mentioned that the code worked previously, it definitely leads me to believe that something has changed in either the `motor.py`, `robot.py`, or the `basic_motion.ipynb` files. However, now that you are saying, "the error are just showing up after a program has run, and not during so its not breaking the system anymore," I am a little confused to how that method is getting called/implemented. Is this for a custom code/script and not one of the example notebooks provided for the Jetbot?

VegardIversen, 2020-09-24:
both custom and example notebook gets the error

santaimpersonator, 2020-09-24:
I recently updated SparkFun's fork; there was an incorrect class name in that method:
https://github.com/sparkfun/jetbot/commit/1453860b99149a89b6150567e75fd656748eb548

santaimpersonator, 2020-10-09:
@VegardIversen Are you still getting the error message?

VegardIversen, 2020-10-09:
> @VegardIversen Are you still getting the error message?

Yep

VegardIversen, 2020-10-09:
@santaimpersonator yep

santaimpersonator, 2020-10-09:
Have you pulled the fork or modified the file directly as mentioned in the commit (https://github.com/sparkfun/jetbot/commit/1453860b99149a89b6150567e75fd656748eb548)? It should resolve the issue.

VegardIversen, 2020-10-09:
No, but I dont have access to the robot anymore, but i guess that would fix it. I have no way of testing it now..

santaimpersonator, 2020-10-09:
Fair enough. In that case, would you mind closing this issue for now.

If you ever get a hold of the robot again and run into the issue, you can reopen this issue. I'd be happy to assist you again.

seeeheng, 2021-02-04:
> Have you pulled the fork or modified the file directly as mentioned in the commit ([sparkfun@1453860](https://github.com/sparkfun/jetbot/commit/1453860b99149a89b6150567e75fd656748eb548))? It should resolve the issue.

Hi @santaimpersonator -

I had the same error message, and I can confirm that changing line 60's `self._motor.disable()` to `self._driver.disable()` resolves the issue. Seems like it was a simple attribute naming error haha.

Thanks for the help!

santaimpersonator, 2021-02-04:
@seeeheng Thanks for the feedback.
@jaybdub I think this issue is resolved and can be closed.

closed 2022-07-15 by jaybdub