NVIDIA-AI-IOT/jetbot #232: Basic Motion Works Until I Attempt to "Link Motors to Triatlets"

WoulfHound, 2020-07-28:
Hi,

I have the SparkFun JetBot AI Kit, KIT-15365. Which according to their website has been retired.

I have been following the Basic Motion tutorial and now I am stuck. The lines of code that are run at the beginning of the tutorial function properly and the robot moves. Such as:
[image]
and
[image]
Once I get to the blocks of code under the "Link Motors to Triatlets" The robot remains unresponsive when the widgets are created and used after linking them to the traitlets.
[image]
The sliders do indeed appear and do indeed seem to function, however, they do not seem to be liked to the motors even after running both these two blocks of code.
Likewise, these to blocks of code have no response from the robot: 
[image]
Not sure what I am missing here, please help.

santaimpersonator, 2020-08-05:
Hey there, you seem to be having an issue similar to issue #227. Try following the steps listed there.

santaimpersonator, 2020-10-09:
@WoulfHound If this issue has been resolved, please close the issue.

closed 2022-07-15 by jaybdub