NVIDIA-AI-IOT/jetbot #233: [SOLUTION] Road Following not working properly Problem Solution

Jeff-kookmin, 2020-08-08:
Hi guys. It is my first time talking to people on githubs. As I have got lots of help from github, I would like to share my case how I made it to work. 

When I followed the instructions in Road following of Jetbot curriculum ,it couldn't get information as fast as the robot goes. I found out that the process of robot is little bit slow. Then I tried to go slooooowly and another problem has occurred. The motor provided was not strong enough to go slowly. (not enough Toque problem)

And we tried to go and stop to manage the speed. (if there were imu in it, we could control the robot with the pid or other feedback control but ,,, no) And here below is the code for the 'go and stop' . 

Anyone who wants to check if the data collected works fine or not, you can try to move the robot really slow with this code, and check for the deep-learning trained model.

```
angle = 0.0
angle_last = 0.0
count_motor = 0
def execute(change):
    global angle, angle_last, count_motor
    image = change['new']
    xy = model(preprocess(image)).detach().float().cpu().numpy().flatten()
    x = xy[0]
    y = (0.5 - xy[1]) / 2.0
    
    x_slider.value = x
    y_slider.value = y
    
    speed_slider.value = speed_gain_slider.value
    
    angle = np.arctan2(x, y)
    pid = angle * steering_gain_slider.value + (angle - angle_last) * steering_dgain_slider.value
    angle_last = angle
    
    steering_slider.value = pid + steering_bias_slider.value
   
    if count_motor >2:
        robot.left_motor.value = max(min(speed_slider.value + steering_slider.value, 1.0)*0.95, 0.0)
        robot.right_motor.value = max(min(speed_slider.value - steering_slider.value, 1.0), 0.0)
        count_motor = 0
    else :
        robot.left_motor.value = 0
        robot.right_motor.value = 0
        count_motor = count_motor + 1

        
execute({'new': camera.value})
```
Gain value :
```

speed_gain_slider = ipywidgets.FloatSlider(min=0.25, max=1.0, step=0.01, description='speed gain')
steering_gain_slider = ipywidgets.FloatSlider(min=0.0, max=1.0, step=0.01, value=0.05, description='steering gain')
steering_dgain_slider = ipywidgets.FloatSlider(min=0.0, max=0.5, step=0.001, value=0.0, description='steering kd')
steering_bias_slider = ipywidgets.FloatSlider(min=-0.3, max=0.3, step=0.01, value=0.0, description='steering bias')
```

jaybdub, 2020-08-08:
Hi Jeff-kookmin,

Thanks for reaching out and sharing this!

Glad to hear you were able to get something working.  Hopefully others can benefit from your experience.

Feel free to reach out if you run into any issues in the future or have any questions.

Best,
John

abuelgasimsaadeldin, 2020-08-18:
Hi Jeff,

I am not achieving expected results with my road following and you mentioned that "to check if the data collected works fine or not, you can try to move the robot really slow with this code, and check for the deep-learning trained model" i'm really new to this i'm sorry but could you tell me where I should copy this code to and how I can see the deep-learning trained model reaction?

Regards.

abuelgasimsaadeldin, 2020-08-18:
Hi all,

I haven't been able to achieve the expected results for my road following even after updating to this version of the code. My robot has a slight left deviation could that be the problem (If yes is there any way to fix this?) or is the deviation normal to have?

Regards.

Jeff-kookmin, 2020-08-18:
Can you tell me where you are on? I mean I want know your situation. Did
you follow the jupyter?

Jeff-kookmin, 2020-08-18:
If it is fine with you, I could help you by facetalk. If you used jupyter
for jetbot, the code will be helpful. But if you are dealing the jetbot
with ROS, I am not sure i could be helpful. Thanks.

abuelgasimsaadeldin, 2020-08-18:
Hi Jeff,

I have done the data collection and have trained my model, however when testing it on the live demo using speed gain of 0.3 (which was the minimum in order for it to run) the robot runs quite fast and does not respond at all as expected. It is also worth mentioning that when running the basic motion in my Jupyter notebook I do experience a slight left deviation when I try for example forward command. 

I am using Jupyter Notebook codes and would appreciate a facetalk. Thanks

Jeff-kookmin, 2020-08-18:
Do you have iphone, facetime? I would like to share my experience with you.
Mine is jeffkimkor@icloud.com

abuelgasimsaadeldin, 2020-08-18:
I'm actually an android user but I do have a close friend working with me using an iphone. Sure will it be possible to have it tomorrow by any chance? I'm located in Malaysia and work during office hours only. 
This is my email ashaikh.saad.as@gmail.com we can keep in touch there.

Regards.

Jeff-kookmin, 2020-08-18:
Cool. What time zone are you in? i am using the same one with Tokyo.

abuelgasimsaadeldin, 2020-08-18:
Cool, we are just an hour behind Tokyo timing. Office hours here is usually from 9am to 5pm.

Jeff-kookmin, 2020-08-18:
Then tell me when you are ready with this email. I would be happy to help
you with. 😊

abuelgasimsaadeldin, 2020-08-18:
Cool, I have just sent an email to you at jeffkimkor@icloud.com

Jeff-kookmin, 2020-08-19:
https://saint-swithins-day.tistory.com/m/44

abuelgasimsaadeldin, 2020-08-19:
Hi, I just wanted to say thank you very much for your time, it was very
nice speaking with you and I appreciate it alot. Thanks!

Jeff-kookmin, 2020-08-19:
I am really happy to hear that. I hope to keep in touch with you in the
future. Good luck with your researches!

Jeff-kookmin, 2020-08-19:
My instagram id is kim.jeff. Please let me know when you get to come to
south Korea.

abuelgasimsaadeldin, 2020-08-19:
Haha thanks will let you know if anything comes up. And if you plan any
trip to Malaysia please do let me know, we can keep in touch.

Thanks and Best Regards,
Abu

closed 2022-07-15 by jaybdub

David-VTUK, 2023-03-15:
@tomMEM - have you deleted your tweaked script?