NVIDIA-AI-IOT/jetbot #239: bluethooth speaker doesnt work when called in Jupyterlab terminal

VegardIversen, 2020-08-16:
Hi,
I have a bluetooth speaker connected to my jetbot. When I run a python program in terminal on the jetson nano that uses the speaker it works with no problem. But when I run the same program from the terminal in jupyterlab (from the IP on the display), its just silent. Seems to be that jupyter are lacking access to use the bluetooth speaker or something, is there a fix for this?

Thank you in advance

tokk-nv, 2020-08-25:
Hi VegardIversen,

Just to check, did you run the Python code on a notebook in Jupyter Lab?
How about if you open a Terminal in Jupyter Lab and run the python program?

VegardIversen, 2020-08-25:
Hi @tokk-nv 
I did run it via the terminal in jupyter lab, and its not working...

jaybdub, 2020-08-27:
Hi Vegardlversen,

Strange indeeed!  My understanding is that Jupyter is simply running a python kernel in the background.

Perhaps there is some issue with permissions, but I'm not sure at what level.

Are you able to share the exact commands you ran on each terminal, as well as perhaps the code (if possible).

Best,
John

VegardIversen, 2020-09-23:
Hi John,
Sorry for late reply. I tested with many different audio libraries as playsound, sounddevice, pyaudio etc that works when i run it via terminal on the jetson nano but not when used via terminal in jupyterlab. example code
```
from gtts import gTTS
import playsound
def speak(text, language='en'):
    tts = gTTS(text=text, lang=language)
    filename = 'voice1.mp3'
    tts.save(filename)
    playsound.playsound(filename)
```

closed 2022-07-15 by jaybdub