NVIDIA-AI-IOT/jetbot #241: High CPU usage by jetbot_stats.service

tomMEM, 2020-08-20:
Hello, I use a jetbot system (last image) without the service display and recognized that the jetbot_stats.service is creating a high CPU usage generating new process numbers. The process is coming and going, so it is difficult to spot. Since PID is changing all the time it is not possible to use 'kill or pkill'. The CPU usage remains high all the time (between 40-90%) using the $ top to monitor. 
Once the service has been removed/stopped the CPU usage became about 4% and overall performance of Ubuntu increased. 

In addition, the object recognition time lag is now about one second in the past. 

Solution:
#check if jetbot_stats.service is in the folder $ cd /etc/systemd/system/
$ ls
#stop service
systemctl stop jetbot_stats             #start service in case: systemctl start jebot_stats
#remove from starting at boot up
systemctl disable jetbot_stats
#delete the file
rm /etc/systemd/system/jetbot_stats

#if the service is required again the follow using "SD card from scratch" point 10.

tokk-nv, 2020-08-25:
Hi tomMEM,

Thank you for the detailed report, especially with the CPU usage data.
I have seen the issue and realized this is an issue especially when the OLED display is not connected.

I filed a Pull Request (#243) which is to introduce some error checking.
It would be great if you can take a look at what I did and see if this is effective in your scenario.

tomMEM, 2020-08-26:
Hello Tokk, thank you for the modifications.
Activation of the old service using old files increased CPU load. 

Doing cd jetbot, and git pull, and sudo python3 setup.py install
replaced the service and the CPU load disappeared, the CPU usage remained so also after reboot. 
The service is sitting now there with a stable PID. All the best.

closed 2020-08-27 by tomMEM