NVIDIA-AI-IOT/jetbot #248: Cannot Run Object_Following

spy-maker, 2020-08-27:
I tried to run object_following using jupyterLab but in the very first step I encounter issue and it is not working. 

Issue I encountered:
1) pybind 11:: init(): factory function returned nullptr
2) name 'model' is not defined
3) name 'detection' is not defined 
4) [Errno 2] No such file or directory: '../collision_avoidance/best_model.pth'
5) name 'preprocess' is not defined 
6) name 'robot' is not defined

jaybdub, 2020-08-27:
Hi spy-maker,

Thanks for reaching out!

Hmm.  Are you using the JetBot SD card image, or did you install by some other means?

Do you mind sharing more details on the steps you took to produce these errors.  For example, like which function calls they occur on.

Please let me know if you have any questions.

Best,
John

spy-maker, 2020-08-27:
Hi John,
Thanks for getting back to me.
I am currently using yahboom jetbot sd card new image. I tried to run the notebook program using jupyterLab and only basic_motion.ipynb is working without issue. So you mean this issue is because of different sd card image I am currently working on is it?

jaybdub, 2020-08-27:
I'm not sure exactly what the issues is, but understanding the system configuration often helps eliminate some variables :).  

Hmm, first it would be good to see if you can acquire an image from the camera.

Just to verify if this is working what happens if you copy the following block of code and execute it in a new Jupyter notebook?

```python
from jetbot.camera import Camera
import matplotlib.pyplot as plt

camera = Camera.instance(width=224, height=224)
image = camera.value

plt.imshow(image)
plt.show()
```

Do you see an image?

Best,
John

spy-maker, 2020-08-28:
> I'm not sure exactly what the issues is, but understanding the system configuration often helps eliminate some variables :).
>
> Hmm, first it would be good to see if you can acquire an image from [...]

Nope, I am getting No module named 'matplotlib'

abulnk, 2020-09-24:
Hi John,

I have installed Matplotlib using 
sudo apt-get install python3-matplotlib 

In Jupyter I get this 

<matplotlib.figure.Figure at 0x7f9c3e97f0>
but no image. Am i supposed to see an image?

closed 2021-03-15 by spy-maker