NVIDIA-AI-IOT/jetbot #255: Support Jetpack 4.4??

vitdao, 2020-08-28:
I want to combine Jetbot and Jetson Inference for my project. Required Tensor RT7, which is minimum Tensor RT version that support loading SSD-Mobile net via ONNX. How can i combine that? Thanks.

tokk-nv, 2020-09-01:
Hi vitdao,
To run JetBot onJetPack 4.4, you may be able to start from this script to setup a SD card from scratch.
https://github.com/NVIDIA-AI-IOT/jetbot/blob/jetpack_4.4/scripts/create-sdcard-image-from-scratch.sh

vitdao, 2020-09-02:
Thanks!! @tokk-nv

closed 2020-09-12 by vitdao

abulnk, 2020-09-12:
Dear tokk-nv,

Thanks for the script. I have used Jetpack 4.4 but it did not install torchvison.  It gave me errors. 

*********************************************************************************************
/usr/local/lib/python3.6/dist-packages/torch/include/ATen/ParallelOpenMP.h:84:0: **warning: ignoring #pragma omp parallel [-Wunknown-pragmas]**
 #pragma omp parallel for if ((end - begin) >= grain_size)
 
In file included from /home/jetbot/Downloads/vision/torchvision/csrc/vision.cpp:14:0:
/home/jetbot/Downloads/vision/torchvision/csrc/ROIAlign.h: In function ‘at::Tensor ROIAlign_autocast(const at::Tensor&, const at::Tensor&, double, int64_t, int64_t, int64_t, bool)’:
/home/ee416/Downloads/vision/torchvision/csrc/ROIAlign.h:52:28: **error: ‘cached_cast’ is not a member of ‘at::autocast’**
              at::autocast::cached_cast(at::kFloat, input),
                            ^~~~~~~~~~~
/home/jetbot/Downloads/vision/torchvision/csrc/ROIAlign.h:53:28: **error: ‘cached_cast’ is not a member of ‘at::autocast’**
              at::autocast::cached_cast(at::kFloat, rois),
                            ^~~~~~~~~~~
In file included from /home/jetbot/Downloads/vision/torchvision/csrc/vision.cpp:17:0:
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h: In function ‘at::Tensor nms_autocast(const at::Tensor&, const at::Tensor&, double)’:
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h:31:21: **error: ‘cached_cast’ is not a member of ‘at::autocast’**
       at::autocast::cached_cast(at::kFloat, dets),
                     ^~~~~~~~~~~
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h:32:21: **error: ‘cached_cast’ is not a member of ‘at::autocast’**
       at::autocast::cached_cast(at::kFloat, scores),
                     ^~~~~~~~~~~
**error: command 'aarch64-linux-gnu-gcc' failed with exit status 1**
*********************************************************************************************

Also, I encountered error trying to install jupyterlab. It needed nodejs>=10.0.0
I used the following link to resolved that:

[https://computingforgeeks.com/installing-node-js-10-lts-on-ubuntu-18-04-16-04-debian-9/](https://computingforgeeks.com/installing-node-js-10-lts-on-ubuntu-18-04-16-04-debian-9/)

I was wondering why I was getting such errors. The rest of the script seems correct. Unfortunately.  I was unable to use "ssd_mobilenet_v2_coco.engine" version v.04 for the object following example of jetbot, because it does not have torchvison. 

The error message is 
**ModuleNotFoundError: No module named 'torchvision'**

I do not want to proceed without check with you or others who may have seen these issues.

By the way, I am using **Sparkfun's jetbot modification files** from 

[https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/all#5-examples-](https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/all#5-examples-)

I have Sparkfun's jetbot with their qwiic perripherals.

Thanks in advance for any help you can provide

Abulnk