NVIDIA-AI-IOT/jetbot #257: Pioled screen not working on my jetbot

shanunrandev123, 2020-08-30:
im working on jetson TX2 and building a Jetbot,but the pioled screen is not showing the ip address,i have checked all the hardware connections

jaybdub, 2020-09-01:
Hi @shanunrandev123 ,

Thanks for reaching out!

I haven't personally tested the PiOLED on the Jetson TX2 yet.  Perhaps there was an issue with permissions.

Since you're not starting from the base JetBot image, do you mind sharing what steps you've taken to configure your system and run the IP address display script?

Best,
John

shanunrandev123, 2020-09-01:
Well, firstly I flashed that latest jetbot image on my sd card using etcher, everything was working well,the oled was showing the ip address but suddenly the voltage tripped and internet connection got lost and there was no display.I tried to Reboot the system again but failed.,I tried to boot Jetson nano again w/o connecting the motor driver but still there wer no results

jaybdub, 2020-09-01:
Hi shanunrandev123,

Just to clarify, are you using a Jetson TX2, or did you mean Jetson Nano in your original post?

I'm sorry to hear the issue, hopefully nothing was broken.  The first thing I would try in this scenario would be to see if it works from wall power and monitor, without anything connected  (even the PiOLED), from a standard Jetson SD card image.  Once we've verified the Jetson Nano is working as usual we could move forward addressing the JetBot specific components.  Be sure to apply 5V when applying to the Jetson Nano directly (I believe some 3rd party kits include 12V chargers for the robot breakout board, which regulates down to 5V, so don't apply that directly to Jetson if that is the case for you).

Please let me know if this helps, or if I'm perhaps misunderstanding the situation.

Best,
John

closed 2022-07-15 by jaybdub