NVIDIA-AI-IOT/jetbot #262: Camera Blank - using sep 2020 clone

tomMEM, 2020-09-04:
Hello, I lost my SD installation, so I started with the last SD-Image and a fresh clone of jetbot again.

Unfortunately the Camera gives blank (empty) images which lead to OpenCV errors - "empty images cannot be processed".
I replaced the very last Camera.py with an old Camera.py (beginning august 2020) and scripts work again.

Please check how to proceed. The camera settings are now in opencv_gst_camera.py, but there seems to be no links to it
Thanks. T

jaybdub, 2020-09-04:
To answer your question around passing parameters using current state of master you can do

```python
from jetbot.camera.opencv_gst_camera import OpenCvGstCamera

camera = OpenCvGstCamera(width=224, height=224)
```

tomMEM, 2020-09-05:
Hello John, thank you for the fast response.
Adding the “from jetbot.camera.opencv_gst_camera import OpenCvGstCamera” and “camera=....” into the cell of the first Camera call of any script worked nicely.

Using git checkout v0.4.0 also helped to resolve the camera issue, however, I had to delete, re-clone the repository and git checkout v0.4.0, because I modified the camera.py before, and I am not sure how to handle the git stash tool.

Thanks again. Best. T

jaybdub, 2020-09-05:
Nice glad to hear it's working for you!

FYI we actually just patched master so you can go back to 

``camera = Camera(width, height)``

Also, it won't be needed here, but in case it helps you in the future, I believe the flow for [git stash](https://git-scm.com/docs/git-stash) is

1. ``git stash`` to undo your changes in your working directory since the last commit and store them in the background
2. ``git checkout ...`` to switch working directory to some other commit
3. ``git stash pop`` to pull your changes out of storage and apply them to the current working directory

This wont do any commits to the repository.  If you test it out, calling ``git status`` before/after each command should help demonstrate what's happening.  

Please let me know if you run into any other issues.

Best,
John

tomMEM, 2020-09-05:
Hello John, thank you for the stash info that might become handy.
The last repository (06092020) is working.
I had problems to get back the old bot performance. 

It looks like the SD image does not allow to use "$sudo jetson_clocks" to switch to Max power, even the fan goes to Max speed, which gave the wrong impression of all four CPUs working, but actually only the CPU frequency goes up. Sure, it is written at many places, but it is easy to miss, and somehow before jetson_clocks was sufficient.

Using $sudo jtop that is preinstalled on the SD image, Max power can be activated ($sudo nvpmodel -m 0  for 10 W.)
I am not sure if jetson_clocks controller need to be activated too.
Time lag of trt roadfollowing display is back to less than a second.
Thanks again.
Best. T

jaybdub, 2020-09-08:
Awesome, thanks for the information.  It may help for us to document these power configuration details more thoroughly.

Out of curiosity, are you using the open source JetBot?  If so which WiFi module are you using, the M2 module or a USB dongle? 

Best,
John

tomMEM, 2020-09-09:
Hello John, the bot system is a buy-by-pieces, however, recommended nano hardware was chosen.
The Wifi hardware is based on an Intel 8265/8275 (ver 78) installed at M2. 
The laptop is connected to the bot over an Android phone using the Hotspot function.
However, regarding time lag, it seems not to make a difference, which type of the three connections is chosen, at least at the level of visual perception. 
Best.
T

closed 2020-10-11 by tomMEM