NVIDIA-AI-IOT/jetbot #268: Support Jetpack 4.4?? #255 -- comments

abulnk, 2020-09-12:
Dear tokk-nv,

Thanks for the script. I have used Jetpack 4.4 but it did not install torchvison. It gave me errors.

/usr/local/lib/python3.6/dist-packages/torch/include/ATen/ParallelOpenMP.h:84:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
#pragma omp parallel for if ((end - begin) >= grain_size)

In file included from /home/jetbot/Downloads/vision/torchvision/csrc/vision.cpp:14:0:
/home/jetbot/Downloads/vision/torchvision/csrc/ROIAlign.h: In function ‘at::Tensor ROIAlign_autocast(const at::Tensor&, const at::Tensor&, double, int64_t, int64_t, int64_t, bool)’:
/home/ee416/Downloads/vision/torchvision/csrc/ROIAlign.h:52:28: error: ‘cached_cast’ is not a member of ‘at::autocast’
at::autocast::cached_cast(at::kFloat, input),
^~~~~~~~~~~
/home/jetbot/Downloads/vision/torchvision/csrc/ROIAlign.h:53:28: error: ‘cached_cast’ is not a member of ‘at::autocast’
at::autocast::cached_cast(at::kFloat, rois),
^~~~~~~~~~~
In file included from /home/jetbot/Downloads/vision/torchvision/csrc/vision.cpp:17:0:
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h: In function ‘at::Tensor nms_autocast(const at::Tensor&, const at::Tensor&, double)’:
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h:31:21: error: ‘cached_cast’ is not a member of ‘at::autocast’
at::autocast::cached_cast(at::kFloat, dets),
^~~~~~~~~~~
/home/jetbot/Downloads/vision/torchvision/csrc/nms.h:32:21: error: ‘cached_cast’ is not a member of ‘at::autocast’
at::autocast::cached_cast(at::kFloat, scores),
^~~~~~~~~~~
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

Also, I encountered error trying to install jupyterlab. It needed nodejs>=10.0.0
I used the following link to resolved that:

https://computingforgeeks.com/installing-node-js-10-lts-on-ubuntu-18-04-16-04-debian-9/

I was wondering why I was getting such errors. The rest of the script seems correct. Unfortunately. I was unable to use "ssd_mobilenet_v2_coco.engine" version v.04 for the object following example of jetbot, because it does not have torchvison.

The error message is
ModuleNotFoundError: No module named 'torchvision'

I do not want to proceed without checking with you or others who may have seen these issues.

By the way, I am using Sparkfun's jetbot modification files from

https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/all#5-examples-

I have Sparkfun's jetbot with their qwiic perripherals.

Thanks in advance for any help you can provide

Abulnk

abulnk, 2020-09-12:
I manged to install pytorch and torchvision using

[https://ngc.nvidia.com/catalog/containers/nvidia:l4t-pytorch](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-pytorch)

Now, back to square one. 

Compute detections on single camera image
from jetbot import ObjectDetector
​
model = ObjectDetector('ssd_mobilenet_v2_coco.engine')
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'

I guess Jetpack 4.4 cannot solve the "Object_following" live demo. I am getting ready to downgrade the jetpack to version 3 with an older version of TensoRT ([link](https://github.com/NVIDIA-AI-IOT/jetbot/issues/126)), which is rather not desirable.

Thanks.

abulnk

vitdao, 2020-09-15:
Hi @abulnk !
You should follow this step:
I hope you solve the problems.

# Install the pre-built TensorFlow pip wheel
sudo apt-get update
sudo apt-get install -y libhdf5-serial-dev hdf5-tools libhdf5-dev zlib1g-dev zip libjpeg8-dev liblapack-dev libblas-dev gfortran
sudo apt-get install -y python3-pip
sudo pip3 install -U pip testresources setuptools numpy==1.16.1 future==0.17.1 mock==3.0.5 h5py==2.9.0 keras_preprocessing==1.0.5 keras_applications==1.0.8 gast==0.2.2 futures protobuf pybind11
# TF-1.15
sudo pip3 install --pre --extra-index-url https://developer.download.nvidia.com/compute/redist/jp/v44 'tensorflow<2'

# Install the pre-built PyTorch pip wheel 
cd
wget -N https://nvidia.box.com/shared/static/yr6sjswn25z7oankw8zy1roow9cy5ur1.whl -O torch-1.6.0rc2-cp36-cp36m-linux_aarch64.whl
sudo apt-get install -y python3-pip libopenblas-base libopenmpi-dev 
sudo -H pip3 install Cython
sudo -H pip3 install numpy torch-1.6.0rc2-cp36-cp36m-linux_aarch64.whl 

# Install torchvision package
cd
git clone https://github.com/pytorch/vision
cd vision
#git checkout v0.4.0
sudo -H python3 setup.py install

# Install torch2trt
cd $HOME
git clone https://github.com/NVIDIA-AI-IOT/torch2trt
cd torch2trt
sudo python3 setup.py install

# Install traitlets (master, to support the unlink() method)
sudo python3 -m pip install git+https://github.com/ipython/traitlets@dead2b8cdde5913572254cf6dc70b5a6065b86f8

# Install jupyter lab
sudo apt install -y curl
curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -
sudo apt install -y nodejs libffi-dev 
sudo -H pip3 install jupyter jupyterlab
sudo -H jupyter labextension install @jupyter-widgets/jupyterlab-manager

jupyter lab --generate-config
python3 -c "from notebook.auth.security import set_password; set_password('$password', '$HOME/.jupyter/jupyter_notebook_config.json')"

# fix for permission error
sudo chown -R jetbot:jetbot ~/.local/share/

abulnk, 2020-09-17:
Hi vitdao,

Thanks for responding to my issue. I have used the .sh file by **tokk-nv** from the following link.

https://github.com/NVIDIA-AI-IOT/jetbot/blob/jetpack_4.4/scripts/create-sdcard-image-from-scratch.sh

I was able to install **torchvision** and **pytorch** with some additional steps.

I will definitely try your steps one more time. I will definitely let you know if I was successfull.

Thanks.

abulnk

abulnk, 2020-09-17:
Hi vitado,

I am using jetso-nano-developer-kit-sd-card-image.zip that has the image **sd-blob-b01**

### **Install torchvision package**

cd
git clone https://github.com/pytorch/vision
cd vision
#git checkout v0.4.0
sudo -H python3 setup.py install

This is where I get the same error I mentioned earlier.

abulnk

tokk-nv, 2020-10-14:
Hi abulnk, vitado,
I'm sorry I've been silent, and thank you very much for putting your efforts into making the JetPack 4.4 base setup.

We are coming out with the new JetBot documentation site, [jetbot.org](jetbot.org) (it is a GitHub Pages, so maintained in this same repo).
There, we are introducing a Docker-based software setup method, that is available for JetPack 4.4
http://jetbot.org/master/software_setup/docker.html

Much of the installation script content went into this [Dockerfile](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docker/base/Dockerfile), but it should be very easy to just follow this guide if you already have JetPack 4.4 sd card, and you will be able to run examples in like 30min (depending on how long it takes for pulling Docker container images from Docker Hub).

I need to note that the Object Following example is still only available for JetPack 4.3 based setup. (as noted [here](http://jetbot.org/master/examples/object_following.html))
But I believe securing the base set up everybody can replicate is a good start.

Please let us know how the Docker-based setup works for you or if you run into any issues!

vitdao, 2020-10-16:
> We are coming out with the new JetBot documentation site, jetbot.org (it

I'll try and feedback errors. Thanks so much for supporting me.

tokk-nv, 2020-10-23:
Hi vitdao, and all,

We just released the new JetBot 0.4.1 sdcard image that is based on JetPack 4.4.1 and updated the documentation.
http://jetbot.org/master/software_setup/sd_card.html

I believe this provides the easy way for you to try.
Please have a check.

closed 2022-07-15 by jaybdub