NVIDIA-AI-IOT/jetbot #269: how to write the control program into jetbot separately

MRinel, 2020-09-16:
Hello,  I have written the Jetson Nano image into SD card, without contol program. Now I want to control the movement of jetbot, but I find there is no control program. Is there any solution to write the control program into jetbot separately? 
Thanks!

VegardIversen, 2020-09-23:
Hi,
If I understand you correctly are you missing the jetbot code (like jetbot.robot etc). If that is the case just follow the instruction to clone the jetbot "library". If this is not the case, what kind of motorcontroller are you using? or if you are building from a kit, which kit?

abulnk, 2020-09-24:
> Hello, I have written the Jetson Nano image into SD card, without contol program. Now I want to control the movement of jetbot, but I find there is no control program. Is there any solution to write the control program into jetbot separately?
> Thanks!

MRinel, Which  Jetbot do you own? There are several that are recommended by NVIDIA and you might be able to find  wiki link for yours.

closed 2022-07-15 by jaybdub