NVIDIA-AI-IOT/jetbot #296: SD card image

akira-sasaki, 2020-10-22:
Docker is great, but it would be still nice to have a complete SD card image! It would save a lot of time when conducting a workshop.

tokk-nv, 2020-10-23:
Hi Akira-san,
Thank you for contacting us and sharing your insight!

Yes, we actually just published the new complete SD card images (both for Jetson Nano (4GB) and Jetson Nano 2GB).
https://jetbot.org/master/software_setup/sd_card.html

Those images internally use the same Docker containers configuration, but you get to save the time to pull container images from Docker Hub as they are pre-loaded.

Please let us know if you face any issues.

greenham592, 2020-11-06:
Thank you for posting this link. I have downloaded the image file and did not have any problems with installing. However, when I am in the Junpyter Lab Terminal to write linux commands, it doesn't regonize some the commands, "command not found".  Such as sudo nvpmodel -q or shutdown. 
The other issue is the motors are not working, Camera and LCD do work without issue. However, I am not sure if that is in the image file or not becasue I am using a Waveshare kit. I have tried to use their code as well, but no luck there either.

tokk-nv, 2020-11-09:
Hi greenham592,

You are right, some of the tools like `nvpmodel` and `tegrastats` are not avaialble within a Docker container.
One thing you can do is to SSH back into your Docker host from the container, like running `ssh jetbot@0.0.0.0` in Jupyter Lab terminal.
We are hoping to find some easy way to access the Docker host terminal from the Jupyter Lab that is running in the container, but I hope the above works for now.

For the motor issue on your Waveshare kit, we need to find that out with Waveshare.

greenham592, 2020-11-09:
Thank you, I will give that a try. As far as the motors, they are finally working after reloading the Waveshare branch several times. Not sure where it was hanging up at but seems to be fine now.

tokk-nv, 2020-11-10:
Hi greenham592,

For the SSH, you need to first install `ssh` on your container.
To preinstall `ssh`, I filed this PR https://github.com/NVIDIA-AI-IOT/jetbot/pull/311 .

And for the Waveshare, yes, it seems that the latest board of Waveshare (with two motor driver chip configuration), it requires their motor.py to move the motors.
For that I filed this PR https://github.com/NVIDIA-AI-IOT/jetbot/pull/312 .

tokk-nv, 2020-11-11:
Hi greenham592,

We updated the master branch with the change in motor.py.
We also made the SD card images (v0.4.2).
http://jetbot.org/master/software_setup/sd_card.html

closed 2022-07-15 by jaybdub