NVIDIA-AI-IOT/jetbot #300: ~/Notebooks folder does not exist

villesau, 2020-10-27:
Many places in the documentation points to `~/Notebooks` which does not exist. For example: https://jetbot.org/master/examples/basic_motion.html

Previously it was recommended to rsync the notebooks folder to `~/Notebooks` but that's not anymore the case: https://github.com/NVIDIA-AI-IOT/jetbot/pull/297/files#diff-815e1a7682683e71aa835a98c87fbd5c3c3064677f11c6dfef6a8ef8020ffdf5L71

tokk-nv, 2020-10-30:
Hi villasau,
Thank you for pointing that out.

This is due to us having a different apparent directory structure on the Jupyter Lab with Docker setup.
We could copy the `notebooks` directory inside the `jetbot` repo and put it directly under the `$HOME` directory.

villesau, 2020-10-30:
Yep, that what I did. Just wanted to point out that there's a lot of mentions about `~/Notebooks` folder and it might be confusing to boot up the system for the first time and realize it's not there.

tokk-nv, 2020-11-02:
Thank you,
I made a Pull Request for adding the copying of the notebooks dir in one of the base Dockerfile.
Let's examine if this is the best way to address this issue.

closed 2022-07-15 by jaybdub