NVIDIA-AI-IOT/jetbot #301: docker build commands broken due to missing base image

villesau, 2020-10-27:
Hi,

seems that base image is not published to docker registry which causes e.g display build `./build.sh` to fail.

Would also be good to have instructions on how to customize different components for your own needs. It took a while to understand that with the new version everything runs in docker.

tokk-nv, 2020-10-30:
Hi villesau,
Thank you for letting us know the issue and the suggestions on the instructions.

Do you run `./build.sh` on your JetPack 4.4 environment, or JetPack 4.4.1?
We may need to specify this also, but we don't have the build process for JetPack versions earlier than that.

villesau, 2020-10-30:
It's 4.4.1, the latest image from here: http://jetbot.org/master/software_setup/sd_card.html I got it eventually working by running build for all the images in docker folder. But when fixing display image for my purposes, I didn't need a modified base image, so would have helped a lot to pull it from the registry.

tokk-nv, 2020-11-02:
Please let me confirm if I understand your situation.
So what you did is to do `cd ./docker/display` and `./build.sh` and found it failed. Is this right?

It may make sense to push the jetbot `base` image to the Docker Hub to enable such branch build, but that would add the complexity of managing extra container images.
Currently, the `display` container has a dependency on the jetbot `base` container because it uses `jetbot.apps.stats` module for OLED display module control.
Perhaps, do you think you would find any (potentially better) alternative solution around this?

villesau, 2020-11-02:
I only had changes for the display image at that point that I tried to get working. The actual problem is that Waveshare jetbot uses it's own fork: https://github.com/waveshare/jetbot . If you look at the ~10 latest commits you can see their patches. I was trying to get the display working by forking jetbot repo and merging waveshare changes on top of it: https://github.com/villesau/jetbot BTW, It _might_ make sense to consider making jetbot compatible with their HW out of the box if possible since I believe it's pretty widely used 3rd party jetbot.

So what I did was that I started by trying to get the display working  before motors (baby steps) and tried to build it by following these instructions: https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker/display#build and running `./build.sh` failed. I assumed that it was since the base image is missing from the docker registry. I got it working by building everything with this script: https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docker/build.sh

Unfortunately I don't have any optimal solution in mind. A bit better documentation might be the best compromise after all. Though, having the base image built and in docker hub would make things a bit faster to build. Also having this part separately: https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docker/base/Dockerfile#L116-L120 would perhaps make sense since that's probably which has the changes if you end up building something customised.

davepae, 2020-11-18:
Is there a faster way to change the jupyter notebook password vs. editing dockerfile and then running docker/build.sh?

closed 2022-07-15 by jaybdub