NVIDIA-AI-IOT/jetbot #303: train_model_plot.ipynb graphs not updating

villesau, 2020-10-27:
Looks like there's something wrong with the Bokeh setup. When running the "Train the neural network" step, printing variables work and loss & accuracy layouts renders once, but the actual graph is never drawn. However, if I restart the step, the graph is updated ate least once.

tokk-nv, 2020-10-30:
Hi villesau,
Thank you for trying out the `train_model_plot.ipynb`.

I need to test this supplemental notebook with the latest setup.
Would you help me by sharing some details on your trial?

Did you execute those in your containers? (without `sudo`)
```
sudo pip3 install bokeh
sudo jupyter labextension install @jupyter-widgets/jupyterlab-manager
sudo jupyter labextension install @bokeh/jupyter_bokeh
```

villesau, 2020-10-30:
@tokk-nv Yes, I run them in the same container running jupyter notebook. As an intermediate feedback: Would be good if those would be included in the image since rebooting wipes them otherwise.

tokk-nv, 2020-11-02:
Hi Ville,
With the 4th cell modified as below to reduce the memory consumption (consistent with other training notebooks), I got to complete the training with the live plot on Jetson Nano 2GB.

```
train_loader = torch.utils.data.DataLoader(
    train_dataset,
    batch_size=8,
    shuffle=True,
    num_workers=0
)

test_loader = torch.utils.data.DataLoader(
    test_dataset,
    batch_size=8,
    shuffle=True,
    num_workers=0
)
```

I'm aware you reported the memory issue on the other post, and you managed to solve the issue by the `shm-size`.
While we may want to check how that impacts, would you also check if you can run the plot notebook with the above modification and actually get the live plot?

[image]

Regarding the image, I also think we should pre-install the bokeh on our image, once we verify everything works with the bokeh plot.

villesau, 2020-11-03:
hmm now it seems to work like a charm! I did the changes you mentioned but I highly doubt that it would be the reason, since the numbers below the graph were printed and model was produced. Maybe it was some temporary glitch with either browser connection, jetson nano or docker then. I've restarted the whole system several times since opening the ticket, though I haven't changed any related code. Originally I tried the step several times, also restarting the jupyter kernel but didn't reboot the container or jetson. It just rendered blank charts but not the actual graphs. Anyways, I'll reopen if this reoccurs.

closed 2020-11-03 by villesau