NVIDIA-AI-IOT/jetbot #304: Jetson nano 2G jetpack4.4.1 collision_avoidance and object_following don't work

lzty634158, 2020-10-29:
1. When I test collision_avoidance experiment, I found that the code could only be executed once and could not be executed all the time.
2. When the object_following, the following error is reported in loading model. I think tensorRT version 7.0 under Jetpack4.4.1 is too high to load the model file generated by tensorRT6.0.

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-78c468b59bd4> in <module>
      1 from jetbot import ObjectDetector
      2 
----> 3 model = ObjectDetector('ssd_mobilenet_v2_coco.engine')

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/object_detection.py in __init__(self, engine_path, preprocess_fn)
     26         load_plugins()
     27         self.trt_model = TRTModel(engine_path, input_names=[TRT_INPUT_NAME],
---> 28                                   output_names=[TRT_OUTPUT_NAME, TRT_OUTPUT_NAME + '_1'])
     29         self.preprocess_fn = preprocess_fn
     30 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/tensorrt_model.py in __init__(self, engine_path, input_names, output_names, final_shapes)
     57         with open(engine_path, 'rb') as f:
     58             self.engine = self.runtime.deserialize_cuda_engine(f.read())
---> 59         self.context = self.engine.create_execution_context()
     60 
     61         if input_names is None:

AttributeError: 'NoneType' object has no attribute 'create_execution_context'

I wonder how everyone can solve the jetson Nano 2G version running Jetbot problem?

tokk-nv, 2020-10-30:
Hi lzty634158,
Thank you for sharing your issues.

As to **collsion_avoidance**;
> When I test collision_avoidance experiment, I found that the code could only be executed once and could not be executed all the time.

Does it look like the camera issue?
You need to always call `camera.stop()` after each run, so that you can run the same code next time.
Could this be it?

As to **object_following**;
As we noted in the document, this example is not currently supported on our JetBot version based on JetPack 4.4.1, so this is currently not made possible on Jetson Nano 2GB.
> Please note, the latest object detection engine targets Jetson Nano running software bundled with JetPack 4.3. This engine not be compatible with other versions of JetPack. For this reason, we recommend starting from the JetPack 4.3 JetBot SD Card image, or using the JetPack 4.3 base SD card image if you want to run the object detection example.
http://jetbot.org/master/examples/object_following.html

lzty634158, 2020-10-31:
@tokk-nv  Thank you, collsion_avoidance is ok. I don't know Jetson 2G when I can supported Jetpack4.4.1，2G Jetson nano only has the image of Jetpack4.4.1， There is no jetpack method to return 4.3？ Since we need to support  the previous 4G version of Jetbot, we would like to know when these problems can be solved？Thank you very much！

lzty634158, 2020-11-02:
@tokk-nv As to collsion_avoidance：
 In jetbot-041_nano-2gb-jp441.zip  docker run script  is nomal，but in my self-configured images can only be executed once, and the video does not refresh，I don't know why？

jaybdub, 2020-11-02:
Hi Izty634158,

_Regarding object detection_

The object detection model needs to be rebuilt for each version of JetPack.  This is because the underlying software versions change.  The problem arises when there are breaking changes with the model and the underlying software, which needs to be manually debugged and fixed.  We've hit that point with the latest JetPack (4.4.1), but have not yet been able to provide a fix and updated Object Detection model.  Unfortunately we don't have an exact timeline on this fix,  we may instead explore other object detection models which are more robust to version changes.

_Regarding collision avoidance_

While I'm not certain, this could be related to a camera failure discussed in this issue:

https://github.com/NVIDIA-AI-IOT/jetbot/issues/306

You may be able to follow that discussion for more information / guidance.

Please let me know if this helps or you have any questions.

Best,
John

lzty634158, 2020-11-03:
Hi @jaybdub ，thank you response me！
The collision avoidance model has rebuild.  It should be the camera crash. After one run, the size of the log file /var/log/syslog increases rapidly, sometimes directly to dozens of G space. The error message is as follows:

Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov  3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState:  (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)

closed 2022-07-15 by jaybdub