NVIDIA-AI-IOT/jetbot #306: When camera crashes, it's not possible to reinitialize it anymore

villesau, 2020-10-30:
Previously when the camera had crashed, you could simply run `sudo systemctl restart nvargus-daemon`. However in the new docker setup, this is no longer the case and you need to restart maybe the whole bot. 

Crashing usually looks something like this:

```
RuntimeErrorTraceback (most recent call last)
<ipython-input-2-15ee7b489b74> in <module>
----> 1 camera = Camera()
      2 
      3 image_widget = widgets.Image(format='jpeg', width=224, height=224)
      4 target_widget = widgets.Image(format='jpeg', width=224, height=224)
      5 

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.1-py3.6.egg/jetbot/camera/opencv_gst_camera.py in __init__(self, *args, **kwargs)
     35             self.stop()
     36             raise RuntimeError(
---> 37                 'Could not initialize camera.  Please see error trace.')
     38 
     39         atexit.register(self.stop)

RuntimeError: Could not initialize camera.  Please see error trace.
```

and can happen for what ever reason. It happens rather often so would be important to be able to reset the camera without having to reboot the robot.

In optimal case the camera would not crash at all though. I think that the docker setup just adds a layer of unnecessary complexity and is pretty big source of all kinds of weird issues and unstability. Have you considered moving the functionality back to the host os instead?

lzty634158, 2020-10-31:
You need to always call camera.stop() after each run, so that you can run the same code next time.

villesau, 2020-11-01:
@lzty634158 not possible if the whole notebook fails. Then you are stuck with running camera and can't init it for stopping it. There's probably other corner cases that might prevent you from explicitly stopping the camera.

jaybdub, 2020-11-02:
Hi villesau,

Thanks for reaching out!

We moved to docker to simplify the release and portability of JetBot software, but using docker has definitely presented it's caveats.

Ideally, I hope we can bring the docker experience to a level on-par (or better) than the previous native host experience by addressing the underlying issues.  

One issue is the instability of the camera.  This was an issue previously, but the failure recovery was simpler.  We have an experimental solution, but have not tested it as thoroughly.  If you are able to try it out, I'm interested to hear your feedback.

### Experimental solution

The experimental solution is to run the camera in it's own process / docker container which publishes messages to Jupyter notebooks.  This isolates the camera from any failures inside the Jupyter container.  It also allows multiple notebooks and processes to access the camera.

To try this:

1. Disable running containers

    ```bash
    ./disable.sh
    ```

2.  Re-enable containers (including the camera publisher container)

    ```bash
    JETBOT_CAMERA=zmq_camera ./enable.sh $HOME
    ```
3.  Replace the camera code in notebooks with the following.  It should behave similarily, but will internally subscribe to camera messages from the publisher.
    
    ```bash
    from jetbot.camera.zmq_camera import ZmqCamera

    camera = ZmqCamera()
    ```

Please let me know if this works for you or you run into issues.  Please note, this is an experimental solution and has not received as much testing, so we cannot guarantee this would become the default in case other issues arise.

Best,
John

villesau, 2020-11-02:
Thanks, I'll try that!

Using docker has it pros and cons. From a hobbyist perspective it's mostly pretty frustrating: I just want to validate and tinker my ideas, and not to worry about docker. For that kind of stuff having everything on host would just make things easier. However I understand that for real development docker might make sense, but the basis should still be solid.

jaybdub, 2020-11-02:
Yeah I 100% understand that.  That's often my personal experience with docker as well.  For tinkering, having a native setup is nice.  

For us, from a maintainability perspective, and for the AI beginner, who wont mess with the system setup too much, it may be a decent fit.

Perhaps as we aggregate more patterns for using docker with Jetson, it will become more hacker friendly.  If you feel there is anything missing in the JetBot documentation that would help hobbyists looking to use docker, I'm eager to hear.

Anyways, look forward to hearing how the experimental camera works for you.

Best,
John

lzty634158, 2020-11-03:
Hi @villesau 

It should be the camera crash. After one run, the size of the log file /var/log/syslog increases rapidly, sometimes directly to dozens of G space. The error message is as follows:

Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: SCF: Error InvalidState: Session has suffered a critical failure (in src/api/Session.cpp, function capture(), line 667)
Nov 3 10:02:25 jetbot-2G nvargus-daemon[5043]: (Argus) Error InvalidState: (propagating from src/api/ScfCaptureThread.cpp, function run(), line 109)

Are you in the same situation as me?

tomMEM, 2020-11-03:
Hello, regarding syslog, you could try:
    However, to prevent syslog build-up at the nano (can reach Max SD card space e.g. 35 GiB) deactivate rsyslog service with

    $sudo service rsyslog stop
    $sudo systemctl disable rsyslog.service
    $cd /var/log
    $sudo rm syslog
    $sudo truncate –s 0 *
    $sudo systemctl status rsyslog  

It does not prevent the error but at least little I/O errors do not blow up the SD.
Best, T

lzty634158, 2020-11-03:
Hi @tomMEM 
This is also my one of problems.
Thank you very much!

John-chq, 2020-11-12:
Hi @villesau 
I'm working at nvidia nano jetbot.
I have the same problem with you,bro.
Have you solved this problem? I can't initialize camera.
Please tell me , thanks~

fariz, 2021-02-18:
same here please help

Pranav-India, 2021-04-01:
@fariz try camera.stop() it worked for me.

jlutzwpi, 2021-05-19:
I have the same issue.  I tried the experimental ZmqCamera above.  it gives me a black screen, which i guess is better than crashes and rebooting the nano?  Please help!

closed 2022-07-18 by jaybdub

federista, 2022-09-15:
@Pranav-India brother I run script line by line in visual studio
Adding camera.stop() in the end is not making any difference in removing the error. 
looking forward to hearing from you