NVIDIA-AI-IOT/jetbot #307: Jetbot Container cannot be initiated - JetPack 4.4

Ryan-ZL-Lin, 2020-11-02:
Hi 
 I followed the instruction right here (https://jetbot.org/master/software_setup/docker.html) to run the Jetbot container, however, I cannot successfully run the container of jupyter, here is the log while running the “enable.sh” in jetbot folder (cloned from https://github.com/NVIDIA-AI-IOT/jetbot.git )

23@123-desktop:~/jetbot/docker$ ./enable.sh $HOME
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
1083206b938b4aa9e3515b79ba2e2aea7dcfc07f2303b01bf9ed95c85f0f1f48
docker: invalid reference format.
See ‘docker run --help’.

when I run disable.sh, here is the log I got

123@123-desktop:~/jetbot/docker$ ./disable.sh
**Error response from daemon: No such container: jetbot_camera**
Error: No such container: jetbot_camera
**Error response from daemon: No such container: jetbot_jupyter**
Error: No such container: jetbot_jupyter
jetbot_display
jetbot_display

I also checked the container running status via _sudo docker container ls,_ and here is what is the output:
CONTAINER ID IMAGE COMMAND CREATED STATUS PORTS NAMES
ae731f7750b7 jetbot/jetbot:display-0.4.1-32.4.4 “/bin/sh -c 'python3…” 4 minutes ago Up 4 minutes jetbot_display

it seems that the **jetbot_display** container is initiated, but the **jetbot_jupyter** and j**etbot_camera** are not.
Is there anyone could help identify the possible root causes, please? Thanks.

jaybdub, 2020-11-02:
Hi Ryan-ZL-Lin,

Thanks for reaching out!

It's expected that by default the ``jetbot_camera`` container is not instantiated.  ``enable.sh`` should enable just the jetbot_jupyter and jetbot_display containers.

It is odd that the jetbot_jupyter container is not enabled.  

Do you mind sharing

1.  Which jetson platform are you using?  (Ie: Jetson nano, Jetson nano 2GB)
2.  Which version of JetPack are you using (ie:  4.4, 4.4.1)

Best,
John

Ryan-ZL-Lin, 2020-11-03:
Hi John
Thanks for reviewing this issue. 
I'm using Jetson nano (not 2 GB)

and here is my JetPack version (I guess it mean 4.4)
123@123-desktop:~$ dpkg-query --show nvidia-l4t-core
**nvidia-l4t-core	32.4.4-20201016124427**

Ryan-ZL-Lin, 2020-11-04:
@jaybdub 
Any findings?

I have tried to use SDK Manager to flash my jetson nano and clone the git again.

While executing **_./enable.sh $HOME_**, here is what I got
Restarting docker daemon ...
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Unable to find image 'jetbot/jetbot:display-0.4.1-32.4.4' locally
display-0.4.1-32.4.4: Pulling from jetbot/jetbot
.
.
.
.
Status: Downloaded newer image for jetbot/jetbot:display-0.4.1-32.4.4
66a9a74f3c50f6936de27c5aed13c06135f34511775993e7c6d0391c7bfa21a5
**docker: invalid reference format.**
See 'docker run --help'.

I'm not quite sure where this error "docker: invalid reference format" comes from, do you have any clues?

Ryan-ZL-Lin, 2020-11-04:
Hi @jaybdub 
I also tried to foolw the instruction right here (https://github.com/NVIDIA-AI-IOT/jetbot/wiki/software-setup) to install latest software (step 5), while executing the command **_sudo python3 setup.py install_**, and got error like this

traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from setuptools import setup, find_packages, Extension
ModuleNotFoundError: No module named 'setuptools'

I'm not sure whether it's related to the root cause, but it's another approach that I have tried, do you have any comments?

Ryan-ZL-Lin, 2020-11-06:
I found the root cause and solve it by myself.
Since I used 2 CSI cameras, I should adjust the /jetbot/docker/jupyter/enable.sh in the following way.

change this line 
--device /dev/video* \

to 

--device /dev/video0 \
--device /dev/video1 \

closed 2020-11-06 by Ryan-ZL-Lin

kritch83, 2020-11-16:
Same exact issue... changing the code for me didn't work; but I just unplugged my 2nd camera and it installed/runs fine.  not going to be using the usb on JetBot anyway.  Thanks Ryan-ZL-Lin for figuring it out.

calleliljedahl, 2021-08-18:
Having the exact same issue... I only have one camera though - any tips on what i could try? 
Thanks