NVIDIA-AI-IOT/jetbot #309: Motors Will Not Test-Run in Jupyter-Docker Notebook or ssh terminal session

Vernieface, 2020-11-07:
[I am a NOOB. First time posting an issue. I regret that I have been unable to complete my installation of mike and mkdocs. I did install mike and have a mkdoc.yml file set up, but not yet working or accessible for markup. So I ask your tolerance as I submit documentation that is not in compliance with contributing guidelines.]

As to basic_motion for jetbot in docker, with a new kernel, I am unable to get past robot-Robot(): 
**Basic Motion**...
...To import the Robot class, highlight the cell below and press ctrl + enter or the play icon above. This will execute the code contained in the cell
[1]: from jetbot import robot
##VPL-First, find Adafruit_MotorHAT and "get" it into the right path. Now that we've imported the Robot class we can initialize the class instance as follows.
[2]: pip install Robot
Requirement already satisfied: Robot in /usr/local/lib/python3.6/dist-packages
Note: you may need to restart the kernel to use updated packages.
[3]: pip install Adafruit_MotorHAT
Requirement already satisfied: Adafruit_MotorHAT in /usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg
Requirement already satisfied: Adafruit-GPIO>=0.7 in /usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.4-py3.6.egg (from Adafruit_MotorHAT)
Requirement already satisfied: adafruit-pureio in /usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-1.1.7-py3.6.egg (from Adafruit-GPIO>=0.7->Adafruit_MotorHAT)
Requirement already satisfied: spidev in /usr/local/lib/python3.6/dist-packages/spidev-3.5-py3.6-linux-aarch64.egg (from Adafruit-GPIO>=0.7->Adafruit_MotorHAT)
Note: you may need to restart the kernel to use updated packages.
[4]: def Robot():
  File "<ipython-input-4-16504fa33384>", line 1
    def Robot():
                ^
SyntaxError: unexpected EOF while parsing
[5]: robot = Robot()
NameErrorTraceback (most recent call last)
<ipython-input-5-b418ad9f6ab3> in <module>
----> 1 robot = Robot()
NameError: name 'Robot' is not defined
Commanding the robotI have

In the NVIDIA jetbot terminal I tried to run the basic_motion tests outside Jupiter and Docker. I have the following:
Last login: Sat Nov  7 11:54:35 on ttys000
(base) Verns-iMac-9:~ vernludwig$ ssh vern@10.0.0.135
vern@10.0.0.135's password: 
Welcome to Ubuntu 18.04.5 LTS (GNU/Linux 4.9.140-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.

63 packages can be updated.
31 updates are security updates.

Last login: Sat Nov  7 11:55:00 2020 from 10.0.0.244
vern@vern-desktop:~$ cd jetbot
vern@vern-desktop:~/jetbot$ def Robot()
-bash: syntax error near unexpected token `('
vern@vern-desktop:~/jetbot$ robot=Robot()
-bash: syntax error near unexpected token `('
vern@vern-desktop:~/jetbot$ pip3 install Robot
Collecting Robot
  Downloading https://files.pythonhosted.org/packages/d8/6d/5acc8d030cb22f1aef304197c4e780db65f0208c8e00c62ed8bbafed17b8/robot-20071211.tar.gz
Building wheels for collected packages: Robot
  Running setup.py bdist_wheel for Robot ... done
  Stored in directory: /home/vern/.cache/pip/wheels/a9/d9/2c/d7e602993188fa3406d21ca2e2b2be09ac2a4fe41bbd45b412
Successfully built Robot
Installing collected packages: Robot
Successfully installed Robot-20071211
vern@vern-desktop:~/jetbot$ def Robot
-bash: def: command not found
vern@vern-desktop:~/jetbot$ robot=Robot()
-bash: syntax error near unexpected token `('
vern@vern-desktop:~/jetbot$ robot=Robot()
-bash: syntax error near unexpected token `('
vern@vern-desktop:~/jetbot$ ls
assets        CMakeLists.txt   docker  jetbot      mkdocs.yml  README.md  setup.py
CHANGELOG.md  CONTRIBUTING.md  docs    LICENSE.md  notebooks   scripts
vern@vern-desktop:~/jetbot$ mkdocs serve --dev-addr=0.0.0.0:8000
INFO    -  Building documentation... 
WARNING -  Config value: 'dev_addr'. Warning: The use of the IP address '0.0.0.0' suggests a production environment or the use of a proxy to connect to the MkDocs server. However, the MkDocs' server is intended for local development purposes only. Please use a third party production-ready server instead. 
INFO    -  Cleaning site directory 
INFO    -  Documentation built in 4.40 seconds 
[I 201107 13:33:45 server:335] Serving on http://0.0.0.0:8000
INFO    -  Serving on http://0.0.0.0:8000
[I 201107 13:33:45 handlers:62] Start watching changes
INFO    -  Start watching changes
[W 201107 13:33:54 web:2243] 404 GET /versions.json (10.0.0.244) 13.83ms
WARNING -  404 GET /versions.json (10.0.0.244) 13.83ms
[I 201107 13:33:54 handlers:135] Browser Connected: http://10.0.0.135:8000/
INFO    -  Browser Connected: http://10.0.0.135:8000/

With this information, can you offer some guidance. Thank you.

tokk-nv, 2020-11-11:
Hi Vernieface,
What hardware (vehicle) do you use?
If that is Waveshare JetBot Kit, would you check the latest SD card image?
http://jetbot.org/master/software_setup/sd_card.html

Vernieface, 2020-11-11:
Great to hear from you. 

I use the Sparkfun AI Jetbot kit 2.1 (kit-16417). I flashed jetbot-042_nano-2gb-jp441.zip on the 64GB SD card with Etcher.  

Vernieface 

 

Vern

tokk-nv, 2020-11-12:
HI Vernieface,

SparkFun provides the pre-flashed SD card with their customized image.
https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/4-software-setup-guide-from-nvidia

You cannot use jetbot-042_nano-2gb-jp441.zip as it is not compatible with their hardware.

Vernieface, 2020-11-14:
Chitoku and dusty_nv,

Thank you, Chitoku, for clairifying hardware-software compatibility. The following will not run on the JetBot: rosrun JetBot_ros JetBot_motors.py
Am I running rosrun at the wrong prompt level?

So, I started over, as follows:
Hardware: SparkFun JetBot AI Kit v2.1 (kit-16417)
64 GB Sandisc flashed with: jetson-nano-4gb-jp441-sd-card-image
Process Guide followed: github.com/dusty-nv/jetbot_ros README.md - [ROS nodes and Gazebo model for NVIDIA JetBot with Jetson Nano]

Each step ran through “Testing the JetBot” at rescore
Last login: Fri Nov 13 11:44:25 on ttys000
(base) Verns-iMac-9:~ vernludwig$ ssh vern@10.0.0.135
vern@10.0.0.135's password: 
Permission denied, please try again.
vern@10.0.0.135's password: 
Welcome to Ubuntu 18.04.5 LTS (GNU/Linux 4.9.140-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.

76 packages can be updated.
36 updates are security updates.

Last login: Fri Nov 13 11:44:45 2020 from 10.0.0.244
vern@vern-desktop:~$ roscore
... logging to /home/vern/.ros/log/2d636034-2604-11eb-bfca-00044bec618c/roslaunch-vern-desktop-10122.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vern-desktop:40565/
ros_comm version 1.14.10

SUMMARY
========
PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
NODES
auto-starting new master
process[master]: started with pid [10132]
ROS_MASTER_URI=http://vern-desktop:11311/
setting /run_id to 2d636034-2604-11eb-bfca-00044bec618c
process[rosout-1]: started with pid [10143]
started core service [/rosout]

In new terminal, I always get an IOError:
Last login: Fri Nov 13 18:00:49 on ttys000
ss(base) Verns-iMac-9:~ vernludwig$ ssh vern@10.0.0.135
vern@10.0.0.135's password: 
Welcome to Ubuntu 18.04.5 LTS (GNU/Linux 4.9.140-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.

78 packages can be updated.
36 updates are security updates.

Last login: Fri Nov 13 18:01:16 2020 from 10.0.0.244
vern@vern-desktop:~$ rosrun jetbot_ros jetbot_motors.py
Traceback (most recent call last):
  File "/home/vern/workspace/catkin_ws/src/jetbot_ros/scripts/jetbot_motors.py", line 75, in <module>
    motor_driver = Adafruit_MotorHAT(i2c_bus=1)
  File "/home/vern/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py", line 231, in __init__
    self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
  File "/home/vern/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py", line 59, in __init__
    self.setAllPWM(0, 0)
  File "/home/vern/.local/lib/python2.7/site-packages/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py", line 95, in setAllPWM
    self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
  File "/home/vern/.local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/home/vern/.local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 268, in write_byte_data
    self._device.write(data)
IOError: [Errno 121] Remote I/O error
vern@vern-desktop:~$ 

@vernieface

closed 2022-07-15 by jaybdub