NVIDIA-AI-IOT/jetbot #310: can't find service entries for jupyter-lab & jetbot_stats

huxinda, 2020-11-08:
Hi, newbie here using the latest Jetpack 4.4.1 Jetbot 0.4.1 for 4gb image.

Both jupyter-lab and jetbot_stats display on the PiOLED are running at boot
[image]

However there's no entries in /etc/systemd/system and I can't find them anywhere.

Thanks.

huxinda, 2020-11-08:
looks like it's related to Docker container, which I have no clue of.
[image]

how do I say modify the jetbot_stats code with this Docker container?
Thanks.

tokk-nv, 2020-11-09:
Hi huxinda,
Thank you for reaching out to us.

To modify the `jetbot_stats` Docker container, you need to build the container image.

You can 
1. Modify the code Python code for stat OLED display (https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/jetbot/apps/stats.py)
2. Build containers

    ```
    cd docker
    source ./configure
    ./build.sh
    ./enable.sh
    ```

Would you please try this?

We should probably document this on the documentation page as well, like here?
http://jetbot.org/master/reference/docker_tips.html

jaybdub, 2020-11-10:
Hi @huxinda ,

Thanks for reaching out!

Previous releases of JetBot used systemd to launch the JetBot stats script.  This has been modified, the stats display is now run as an independent docker container, which is managed by the docker daemon.

To view the running docker containers

```bash
sudo docker ps
```

You can view the output of a specific container as follows

```bash
sudo docker logs jetbot_display
```

Please let me know if this helps or you have any further questions.

Best,
John

closed 2022-07-15 by jaybdub