NVIDIA-AI-IOT/jetbot #314: [To be solved] Waveshare JetBot kit's motors do not spin

tokk-nv, 2020-11-10:
### Symptom
When testing the latest SD card image ([jetbot-041_nano-4gb-jp441.zip](http://jetbot.org/master/software_setup/sd_card.html)) on a newly purchased **Waveshare JetBot kit**, the motors do not spin.

### Who is affected by this issue
Waveshare recently introduced a new revision of its expansion board which features different motor driver configurations.
People purchased Waveshare JetBot kit in the recent month (Oct, Nov 2020) have a higher chance of having this new revision kit.
[image]

mmatesic01, 2020-11-11:
It works using 0.4.2 version.

tokk-nv, 2020-11-11:
### Solution
**Use JetBot version `0.4.2`.**
http://jetbot.org/master/software_setup/sd_card.html

### Detail of the issue and the solution
We recently discovered that recent versions of the WaveShare JetBot kit have a motor driver configuration that is incompatible with our open source software.
We update the jetbot software to include a fix for this, made a new release and the SD card images, and updated the documentation acoordigly.

thomtj94, 2020-11-25:
Hey, do you know if there is any way/fix to use the new motor drivers in the Jetbot version with Jetpack 4.3? Because this is the only one where the object detection example should work at least..

tokk-nv, 2020-12-14:
Hi HyperStudio-2020,

You can probably apply this to the old image.
https://github.com/NVIDIA-AI-IOT/jetbot/commit/cbf6f1b5a3285ad3bbb18ec552ed79846d1e2529#diff-e497c529fb992fd1cbf4fcd5a7b0a33d91f45910277556e41b4c3fcc5f5574f8

closed 2022-07-15 by jaybdub