NVIDIA-AI-IOT/jetbot #316: handling the new version of Waveshare

Ryan-ZL-Lin, 2020-11-11:
@tokk-nv san
I noticed that you have committed motor.py 2 days ago with handling the new version of Waveshare board, does that mean I could directly clone this repository here and spin the motor from Waveshare jet to kit?

 Before this commit, I have to replace the whole jetbot repository with Waveshare’s jetbot repository to spin the motor.

tokk-nv, 2020-11-11:
Hi Ryan-ZL-Lin,

> does that mean I could directly clone this repository here and spin the motor from Waveshare jet to kit?

Yes, you can follow this Docker based setup method.
http://jetbot.org/master/software_setup/docker.html

Or you can also download the SD card image.
http://jetbot.org/master/software_setup/sd_card.html

Ryan-ZL-Lin, 2020-11-12:
Thanks @tokk-nv  san, it works and save me a lot of time integrating different version of jetbot repository.

closed 2020-11-12 by Ryan-ZL-Lin