NVIDIA-AI-IOT/jetbot #317: Module Not Found Error

spy-maker, 2020-11-12:
Hi,
I am working on jetpack version 4.4 from scrath when I ran notebook codes I am keep getting No module named 'jetbot' error. Does anyone know how to fix this ?
I also keep getting same error for Adafruit.

jaybdub, 2020-11-16:
Hi spy-maker,

Thanks for reaching out!

This error indicates the jetbot package is not installed.  Do you mind sharing the steps you followed, as well as the commands which produced these errors?

Best,
John

spy-maker, 2020-11-16:
Previously I was using jetpack version 4.3 of another image. Due to the limitations I switched to 4.4 jetpack version. After I download the jetpack image from Nvidia website and used docker to download jetbot. I also downloaded the adafruit modules from online and after the setup it works but I couldn't fix this jetbot module issue. 

So right now I am wondering whether I should try flash the new jetbot image with 4.4 jetpack version instead of using docker. And is there any specific steps to download jetbot modules?

Normally after I flash the image I will setup things such as:
- apt update & upgrade
- download deep learning framework 

I also wanted to check is there any other steps for setup this ?

jaybdub, 2020-11-16:
Hi spy-maker,

FYI, flashing the latest _JetBot_ image actually wraps the docker container.   Because of this, the new JetBot image may not help here.  There has been some frustration with this docker update in the community, which it sounds like your use case resonates with.  Please let me know if you have any feedback on this, or ideas on what you _expect_ to see from the JetBot SD card image.  This will help inform our judgment.

To install on a native configuration, there are a few steps.

1.  Install software dependencies.
    1.  The main one is PyTorch, you can find it here https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-7-0-now-available/72048
    2.  Many other dependencies can be installed by referencing the base docker file https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docker/base/Dockerfile#L72

After the software dependencies are installed, you should be able to import the JetBot library.  To get the JetBot development experience, you will need to configure the system to launch the Jupyter lab server at boot.

Please let me know if this helps or you run into issues installing the software dependencies, and I'm happy to help with instructions for getting the Jupyter lab server running.

Best,
John

spy-maker, 2020-11-17:
Dear John,
I have flashed new jetbot image with jetpack version 4.4.1 directly into the sd card instead of using docker method. Is it possible for you to guide me on how to setup things after I flash new sd card image?

Especially the jetbot modules part. 
Regards,
Karthick

hex0r, 2020-11-18:
Hi everyone,

I have an question regarding jetbot modules and jupyter. I had to change the robot.py due to custom motor driver but now i have the following issue if I execute the basic_motion notebook 

OSErrorTraceback (most recent call last)
<ipython-input-2-b418ad9f6ab3> in <module>
----> 1 robot = Robot()

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.1-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

The error itself is not the problem because I dont use motor_hat but this is not the same code as in my robot.py.
I already deleted the /usr/local/lib/python3.6/dist-packages/jetbot-0.4.1-py3.6.egg/jetbot/robot.py and add the fowling code at the beginning 

import sys
import importlib
sys.path.insert(0, '/home/jetbot/jetbot')
from jetbot import Robot

If is use runipy ./notebooks/basic_motion/basic_motion.ipynb everthing works as expected.
What is the different to jupyter
tried  to reboot, shutdown kernels.. 
and how can jupyter read the  old /usr/local/lib/python3.6/dist-packages/jetbot-0.4.1-py3.6.egg/jetbot/robot.py even it is already deleted.
I am not an python expert, so can anybody give me an hand please??
thx

spy-maker, 2020-11-19:
Hi John,
I tried flashing jetbot image directly instead of using docker and it works well this time. I didn't encounter jetbot module issue this time. Currently, I am setting up this jetpack 4.4.1 version using my customised jetbot. Previously I was working on it using 4.3. So right now, I am setting up this to make it run. I already have the required files needed but I am getting errors. 

I want to get clarify few things:

1) After I flashed the new image I modified pioled display code to make it work. I changed **i2c to 0** and it works well. Later I created display service to make it auto run after every time it boot up. It only display ip address after I enter the password is there any ways to fix this. Do I need to create Jupyter service ?

2) Like I mentioned earlier I already have required modules and drivers needed to run but I dont know how to setup those because I already have those modules installed. Can you help me with this ?
I need to setup adafruit motor, servo, gpio and Rpi.

I am unable to setup and configure those mentioned above and it resulted to errors when I run codes. I am getting **os error error no 121 Remote I/O error**

spy-maker, 2020-12-14:
> Hi everyone,
>
> I have an question regarding jetbot modules and jupyter. I had to change the robot.py due to custom motor driver but now i have the following issue if I execute the basic_motion [...]

Try using the old jetbot module which was used previously for training. I flashed new jetpack version and installed required drivers and for the modules try using the old version. For example I am using jetbot 0.3.0 previously but when I used 0.4.1 I get this issue so try switching back to previous version.

closed 2020-12-14 by spy-maker