NVIDIA-AI-IOT/jetbot #319: how to make changed gstream effective on docker container?

Ryan-ZL-Lin, 2020-11-12:
Hi. @tokk-nv 
I tried to use the new version of jetbot repository after your commit of handling waveshare board.
Because I have 2 CSI cameras installed on jetbot, I need to change the gstream in **jetbot/camera/opencv_gst_camera.py** to use the right camera for the demo.

Here is the updated code. I only added the sesnor-id parameter in the gstream pipeline  but it doesn't reflect on juputer after reboot.

`def _gst_str(self):
        return 'nvarguscamerasrc **sensor-id=1** ! video/x-raw(memory:NVMM), width=%d, height=%d, format=(string)NV12, framerate=(fraction)%d/1 ! nvvidconv ! video/x-raw, width=(int)%d, height=(int)%d, format=(string)BGRx ! videoconvert ! appsink' % (
                self.capture_width, self.capture_height, self.fps, self.width, self.height)`

Is there anyone could help me solve this issue or identify the possibile causes? thanks.
p.s. I do executed **Python3 setup.py install** after I changed the code .

Ryan-ZL-Lin, 2020-11-13:
BTW, do I need to build the docker container again?
I tried to build the docker container via executing jetbot/docker/build.sh, but I got an error like this: 

./build.sh: 3: ./build.sh: [[: not found
./build.sh: 6: ./build.sh: [[: not found
invalid argument "/jetbot:base--" for "-t, --tag" flag: invalid reference format
See 'docker build --help'.
./build.sh: 2: cd: can't cd to models
./build.sh: 3: cd: can't cd to display
./build.sh: 4: cd: can't cd to jupyter
./build.sh: 5: cd: can't cd to camera

jaybdub, 2020-11-16:
Hi Ryan-ZL-Lin,

Thanks for reaching out!

It may be easiest to build the container, using your local JetBot checkout with changes applied.

My guess is you need to call ``configure.sh`` to set the environment variables which are used to assign a tag to the container.

```bash
cd docker
```

Edit ``configure.sh`` to set ``JETBOT_VERSION=custom``, or your branch's name.

```bash
JETBOT_VERSION=custom
```

Source``configure.sh`` to set the environment variables

```bash
source configure.sh
```

You could then call

```bash
./build.sh
```

According to [this script](https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docker/jupyter/build.sh#L3), the container should have a name like

```bash
jetbot/jetbot:jupyter-custom-32.4.4
```

Then, you can run your container by calling

```bash
enable.sh $HOME
```

Please let me know if that helps or you run into any issues.

Best,
John

tomMEM, 2020-11-17:
Hello @Ryan-ZL-Lin, in case of update problems of scripts in "build" in response to "python3 setup.py install" I had to add a "nonsense" row in scripts within the first rows (e.g. #asdfsdf). Only then updates took place. 
Regarding Docker, it seems by definition that changes in scripts within the container will not be permanent, thus a rebuild of the docker container is likely required.

Ryan-ZL-Lin, 2020-11-17:
Thanks @jaybdub and @tomMEM I make it works after your suggestion to re-=build the docker container.

closed 2020-11-17 by Ryan-ZL-Lin