NVIDIA-AI-IOT/jetbot #325: High Pitch Noise when running teleoperation

liamhan0905, 2020-11-23:
It's been a while and I'm coming back to this to work on a project.

When running the code shown below and moving a joystick, I start hearing a very high pitch ringing noise from the speed controller. I don't remember the noise being this audible.. Has anyone had a similar issue recently?

```
from jetbot import Robot
import traitlets

robot = Robot()

left_link = traitlets.dlink((controller.axes[1], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)
right_link = traitlets.dlink((controller.axes[3], 'value'), (robot.right_motor, 'value'), transform=lambda x: -x)
````

tokk-nv, 2020-11-30:
Hi liamhan0905,

When you set some small value other than 0 like 1 to the value and leave it, the motor continues trying to weakly move the motor but does not produce enough power to move the whole vehicle, so it can result in the still robot with motors making noise.

Can you make sure you have 0 for both motor values?

DingoOz, 2020-12-28:
@liamhan0905 the audible noise is from the Pulse Width Modulation frequency that is being passed to the motors from the motor driver(s).

If you are not familiar with PWM, this is a good primer: [https://learn.sparkfun.com/tutorials/pulse-width-modulation/all](https://learn.sparkfun.com/tutorials/pulse-width-modulation/all)

Depending on what hardware you are using it might be caused by the motor driver or by software (or an interaction between them).

The settings to play with are in motor.py at line 40 and 43

closed 2022-07-15 by jaybdub