NVIDIA-AI-IOT/jetbot #326: Where can I find how to set the GPU enables training PC?

thomtj94, 2020-11-23:


tokk-nv, 2020-11-30:
Hello HyperStudeio-2020,

The easiest way to set up a Linux training machine is probably to go with Docker on your Linux PC.
This document walks you through the general setup of Docker on your GPU enabled PC.
https://docs.nvidia.com/deeplearning/frameworks/user-guide/index.html

There is a PyTorch docker container available on NGC, but you may need to install Jupyter environment by yourself.
https://ngc.nvidia.com/catalog/containers/nvidia:pytorch

HyperStudio-2020, 2020-12-02:
Hey!

Thanks for the advice, is there any way I can set this up and train on my
windows PC, I have an RTX 2060, so that's fine for training. But I really
dont know how to train a model, just as the COCO model for example. How can
I get my custom .engine file?

Thanks for reading! Hope to hear from you soon!

With best regards,

Thom

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2022-07-15 by jaybdub