NVIDIA-AI-IOT/jetbot #327: Object following example not working :(

thomtj94, 2020-11-23:
Heu guys,

I can't get this examples (and actally all examples which use trained models) to work.

It gives the error:

name 'model' is not defined

Would be amazing if somebody knows the solution to this

thomtj94, 2020-11-23:
BTW, I am using a Waveshare Jetbot Kit bought in november 2020, installed Jetbot through the newest container method

whmzsu, 2020-12-17:
jetbot.org  had notice about the example:
Please note, the latest object detection engine targets Jetson Nano running software bundled with JetPack 4.3. This engine not be compatible with other versions of JetPack. For this reason, we recommend starting from the JetPack 4.3 JetBot SD Card image, or using the JetPack 4.3 base SD card image if you want to run the object detection example.

aixian-jiang, 2021-03-31:
Hi Hyperstudio, have you solved the problem? I also have such problem. I just bought waveshare jetbot 2 weeks ago, by the way, I am using Jetpack 4.5. 

Anyone knows how to solve it? very apprieciate if could help to solve it!

santaimpersonator, 2021-10-01:
@aixian-jiang You'd have to manually rebuild the engine for Jetpack 4.5. @jaybdub mentions it here: https://github.com/NVIDIA-AI-IOT/jetbot/issues/126#issuecomment-535186884