NVIDIA-AI-IOT/jetbot #329: Road following example issue, kernel gets killed when trying to load model

reinisnud, 2020-11-26:
After a fresh newest SD image version installation, on JupyterLab trying to run demo_build_trt notebook kernel just dies without any error, tried the same lines of code in the terminal - python console and it says "Killed". 

SD image used:
Jetson Nano (4GB)	4.4.1	0.4.2	jetbot-042_nano-4gb-jp441.zip

What this could mean?

PappyMoelling, 2020-11-27:
I got same error. reimaged SD card and tried again - same thing. went back to previous build and works fine. 
Jetson Nano (4GB) | 4.4.1 | 0.4.1 | jetbot-041_nano-4gb-jp441.zip

reinisnud, 2020-11-27:
Thanks, guess will do the same.

closed 2022-07-15 by jaybdub

aranis22, 2023-08-28:
> I got same error. reimaged SD card and tried again - same thing. went back to previous build and works fine. Jetson Nano (4GB) | 4.4.1 | 0.4.1 | jetbot-041_nano-4gb-jp441.zip

How did you do that?