NVIDIA-AI-IOT/jetbot #338: Cannot transfer model for execution to GPU

Jundor-y, 2020-12-14:
Following the train_model_resnet18 notebook, when it says "Finally, we transfer our model for execution on the GPU", the notebook indicates it's busy for a while, then stops as if nothing happened. The index counter to the left of the cell returns to empty (nothing inside the square brackets) and all other information from previously run cells is lost. So if I used import time on a cell above, using time.sleep(1) will result in `NameError: name 'time' is not defined`. So something goes way wrong with the transfer to GPU.

The issue occurs when running these two lines:
```
device = torch.device('cuda')
model = model.to(device)
```

**Software**:
jetbot-042_nano_4gb-jp441
**Hardware**:
Waveshare Jetbot

tomMEM, 2020-12-15:
Hi, it seems to work only with 041_nano-4gb-jp441.img. Similar issue was mentioned before #330, #329 etc. 
For waveshare you need to copy the motor.py from your jetbot-042_nano_4gb-jp441 to move it later jetbot-042_nano_4gb-jp441 (to jetbot/jetbot) followed by copy to /opt/jetbot and run python3 setup.py install from /opt/jetbot (see #320 at the end).

Jundor-y, 2020-12-15:
Ended up using the suggested image: jetbot_JP4.3_JL1.24_PT1.30_0115.img and it works fine without any further adjustments. 
Out of curiosity, since I also have a JetRacer, that's has it's own issues with training data, I would be interested in copying the motor.py and robot.py from the JetRacer image over onto this image. How would your instructions work exactly with the jetbot_JP4.3_JL1.24_PT1.30_0115.img?
The /opt folder doesn't contain the /jetbot folder. Where exactly should the robot.py and motor.py scripts be copied to?

ajjack50n, 2021-01-31:
Hey, I am having the same issue. Is there any update, or do I simply have to rollback to an earlier image?

closed 2022-07-15 by jaybdub