NVIDIA-AI-IOT/jetbot #349: Jetbot Collision Avoidance example failed to load in the trained model

lchakman, 2021-01-03:
Hello. I am using the waveshare Jetbot kit and developing following the example codes. In the I have followed the Jetbot wiki and trained the model on Google Colab using its online resources and GPU. After data_collection and trained_model notebooks , I now have a best_model.pth in hand. I have downloaded it to my desktop and uploaded it to the same directory as the live_demo.ipynb on my Jetbot.

However, in live_demo.ipynb, when I started to run the following cell block to load in the trained weights of the model, 

model.load_state_dict(torch.load('best_model.pth'))

The following error occurs. 

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/usr/lib/python3.6/tarfile.py in nti(s)
    188             s = nts(s, "ascii", "strict")
--> 189             n = int(s.strip() or "0", 8)
    190         except ValueError:

ValueError: invalid literal for int() with base 8: 'sor_v2\nq'

During handling of the above exception, another exception occurred:

InvalidHeaderError                        Traceback (most recent call last)
/usr/lib/python3.6/tarfile.py in next(self)
   2296             try:
-> 2297                 tarinfo = self.tarinfo.fromtarfile(self)
   2298             except EOFHeaderError as e:

/usr/lib/python3.6/tarfile.py in fromtarfile(cls, tarfile)
   1092         buf = tarfile.fileobj.read(BLOCKSIZE)
-> 1093         obj = cls.frombuf(buf, tarfile.encoding, tarfile.errors)
   1094         obj.offset = tarfile.fileobj.tell() - BLOCKSIZE

/usr/lib/python3.6/tarfile.py in frombuf(cls, buf, encoding, errors)
   1034 
-> 1035         chksum = nti(buf[148:156])
   1036         if chksum not in calc_chksums(buf):

/usr/lib/python3.6/tarfile.py in nti(s)
    190         except ValueError:
--> 191             raise InvalidHeaderError("invalid header")
    192     return n

InvalidHeaderError: invalid header

During handling of the above exception, another exception occurred:

ReadError                                 Traceback (most recent call last)
/usr/local/lib/python3.6/dist-packages/torch/serialization.py in _load(f, map_location, pickle_module, **pickle_load_args)
    594         try:
--> 595             return legacy_load(f)
    596         except tarfile.TarError:

/usr/local/lib/python3.6/dist-packages/torch/serialization.py in legacy_load(f)
    505 
--> 506         with closing(tarfile.open(fileobj=f, mode='r:', format=tarfile.PAX_FORMAT)) as tar, \
    507                 mkdtemp() as tmpdir:

/usr/lib/python3.6/tarfile.py in open(cls, name, mode, fileobj, bufsize, **kwargs)
   1588                 raise CompressionError("unknown compression type %r" % comptype)
-> 1589             return func(name, filemode, fileobj, **kwargs)
   1590 

/usr/lib/python3.6/tarfile.py in taropen(cls, name, mode, fileobj, **kwargs)
   1618             raise ValueError("mode must be 'r', 'a', 'w' or 'x'")
-> 1619         return cls(name, mode, fileobj, **kwargs)
   1620 

/usr/lib/python3.6/tarfile.py in __init__(self, name, mode, fileobj, format, tarinfo, dereference, ignore_zeros, encoding, errors, pax_headers, debug, errorlevel, copybufsize)
   1481                 self.firstmember = None
-> 1482                 self.firstmember = self.next()
   1483 

/usr/lib/python3.6/tarfile.py in next(self)
   2308                 elif self.offset == 0:
-> 2309                     raise ReadError(str(e))
   2310             except EmptyHeaderError:

ReadError: invalid header

During handling of the above exception, another exception occurred:

RuntimeError                              Traceback (most recent call last)
<ipython-input-9-97ca0473330f> in <module>
----> 1 model.load_state_dict(torch.load('best_model.pth'))

/usr/local/lib/python3.6/dist-packages/torch/serialization.py in load(f, map_location, pickle_module, **pickle_load_args)
    424         if sys.version_info >= (3, 0) and 'encoding' not in pickle_load_args.keys():
    425             pickle_load_args['encoding'] = 'utf-8'
--> 426         return _load(f, map_location, pickle_module, **pickle_load_args)
    427     finally:
    428         if new_fd:

/usr/local/lib/python3.6/dist-packages/torch/serialization.py in _load(f, map_location, pickle_module, **pickle_load_args)
    597             if _is_zipfile(f):
    598                 # .zip is used for torch.jit.save and will throw an un-pickling error here
--> 599                 raise RuntimeError("{} is a zip archive (did you mean to use torch.jit.load()?)".format(f.name))
    600             # if not a tarfile, reset file offset and proceed
    601             f.seek(0)

RuntimeError: best_model.pth is a zip archive (did you mean to use torch.jit.load()?)
---------------------------------------------------------------------------------

I've tried to replace torch.load with torch.jit.load as suggested by the error, but to no avail as it leads to another error. 

I am a complete newbie to both the JupyterLab environment as well as AI programming. I have searched this issue on both stackoverflow and this github, but I've found no result. It would be great if anyone can help me in fixing this issue as I am now completely stuck by this error that I do not know how to debug or fix.

Thanks a lot for all the help.

AdelineVieusse, 2021-01-04:
Hello,
It seems that the new versions of PyTorch use the zip format by default when generating the .pth file and you need to specify that you want to use the old format (see PyTorch documentation https://pytorch.org/docs/stable/generated/torch.save.html).
I modified the train_model notebook (at the second to last line of the notebook) to state:
torch.save(model.state_dict(), BEST_MODEL_PATH, _use_new_zipfile_serialization=False)
in lieu of 
torch.save(model.state_dict(), BEST_MODEL_PATH)

And it worked; when I loaded the best_model.pth file in the live_demo notebook (it returned <All keys matched successfully> which is a good sign ;-).
I hope this helps.
Adeline.

lchakman, 2021-01-06:
Hi Adeline,

Thanks for the help! The model loaded successfully just like you said after the modification. Thanks a lot!

closed 2021-01-06 by lchakman