NVIDIA-AI-IOT/jetbot #358: "TimeoutError: [Errno 110] Connection timed out" error using Waveshare Jetbot

abuelgasimsaadeldin, 2021-01-12:
Hi guys, After assembling my new Waveshare Jetbot and powering it on the Jetson Nano turns on but nothing is displayed on the OLED display, I checked if i2c is detected using command "sudo i2cdetect -r -y 1" but it takes really long and I don't think that anything is detected. Next when I try to run the command "robot=Robot()" in the Jupyter Notebook I receive the following error message below, can someone please assist me? (ps. I am using the latest Jetbot Image provided by Waveshare)

TimeoutError                              Traceback (most recent call last)
<ipython-input-2-12e166046241> in <module>
      2 import traitlets
      3 
----> 4 robot = Robot()
      5 
      6 left_link = traitlets.dlink((controller.axes[1], 'value'), (robot.left_motor, 'value'), transform=lambda x: -x)

/usr/local/lib/python3.6/dist-packages/jetbot-0.4.0-py3.6.egg/jetbot/robot.py in __init__(self, *args, **kwargs)
     20     def __init__(self, *args, **kwargs):
     21         super(Robot, self).__init__(*args, **kwargs)
---> 22         self.motor_driver = Adafruit_MotorHAT(i2c_bus=self.i2c_bus)
     23         self.left_motor = Motor(self.motor_driver, channel=self.left_motor_channel, alpha=self.left_motor_alpha)
     24         self.right_motor = Motor(self.motor_driver, channel=self.right_motor_channel, alpha=self.right_motor_alpha)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_MotorHAT_Motors.py in __init__(self, addr, freq, i2c, i2c_bus)
    229         self.motors = [ Adafruit_DCMotor(self, m) for m in range(4) ]
    230         self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]
--> 231         self._pwm = PWM(addr, debug=False, i2c=i2c, i2c_bus=i2c_bus)
    232         self._pwm.setPWMFreq(self._frequency)
    233 

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in __init__(self, address, debug, i2c, i2c_bus)
     57         self.i2c = get_i2c_device(address, i2c, i2c_bus)
     58         logger.debug("Reseting PCA9685 MODE1 (without SLEEP) and MODE2")
---> 59         self.setAllPWM(0, 0)
     60         self.i2c.write8(self.__MODE2, self.__OUTDRV)
     61         self.i2c.write8(self.__MODE1, self.__ALLCALL)

/usr/local/lib/python3.6/dist-packages/Adafruit_MotorHAT-1.4.0-py3.6.egg/Adafruit_MotorHAT/Adafruit_PWM_Servo_Driver.py in setAllPWM(self, on, off)
     93     def setAllPWM(self, on, off):
     94         "Sets a all PWM channels"
---> 95         self.i2c.write8(self.__ALL_LED_ON_L, on & 0xFF)
     96         self.i2c.write8(self.__ALL_LED_ON_H, on >> 8)
     97         self.i2c.write8(self.__ALL_LED_OFF_L, off & 0xFF)

/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.4-py3.6.egg/Adafruit_GPIO/I2C.py in write8(self, register, value)
    114         """Write an 8-bit value to the specified register."""
    115         value = value & 0xFF
--> 116         self._bus.write_byte_data(self._address, register, value)
    117         self._logger.debug("Wrote 0x%02X to register 0x%02X",
    118                      value, register)

/usr/local/lib/python3.6/dist-packages/Adafruit_PureIO-1.0.4-py3.6.egg/Adafruit_PureIO/smbus.py in write_byte_data(self, addr, cmd, val)
    266         # Send the data to the device.
    267         self._select_device(addr)
--> 268         self._device.write(data)
    269 
    270     def write_word_data(self, addr, cmd, val):

TimeoutError: [Errno 110] Connection timed out

abuelgasimsaadeldin, 2021-02-05:
I'm still having this problem, any help would be appreciated. Thanks in advance!

marius12233, 2021-02-06:
I'm having the same problem with a waveshare jetbot ai kit. . Any suggestion?

abuelgasimsaadeldin, 2021-02-08:
Hi @marius12233,

Unfortunately no, not at this moment, I have tried literally so many solutions but all were unsuccessful. My only speculation as of right now is that there may be something faulty in the Waveshare Expansion board itself internally (specifically to do with the I2C). If there is anyone who had faced a similar problem to this and had solved the issue it would be very much appreciated. 

Regards,
Abu

marius12233, 2021-02-08:
@abuelgasimsaadeldin I've sent a message to waveshare to understand if it is a known issue. I'm waiting for response..

marius12233, 2021-02-08:
Ok I've solved the issue, at least for me. The problem is the bus 1 of the nano. I've tried to connect to bus 0 pins and now it worked (After changing the i2c_bus default variable in robot.py to 0).

marius12233, 2021-02-18:
> Ok I've solved the issue, at least for me. The problem is the bus 1 of the nano. I've tried to connect to bus 0 pins and now it worked (After changing the i2c_bus default variable in robot.py to 0).

@abuelgasimsaadeldin

abuelgasimsaadeldin, 2021-02-25:
Hi, @marius12233 

Thank you for the suggestion, I really appreciate it, however I did try it out and it did not work, so I ended up contacting Waveshare and got a new Expansion board and this time it works fine.

However this new Expansion Board OLED display does not work, it worked the first time but not anymore, not sure what is wrong with it but its only the address that is not displayed, everything else is working so I am happy.

Thanks again @marius12233