NVIDIA-AI-IOT/jetbot #360: Power Issue

liamhan0905, 2021-01-13:
I'm having a power issue.. I'm using the [power bank](https://www.amazon.com/gp/product/B07H6LB4J4/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1) listed on the hardware part but after 4-5 seconds of turning the power ON, the power will turn back OFF (using micro usb). When I try using an external power with barrel it works fine...

I'm not sure why this is happening. I didn't have this issue before. I updated to the latest jetpack version 4.4.1 just in case but having the same issue. I still have 2/3 of the battery on the power bank. Anyone else have this issue before?

DingoOz, 2021-01-13:
Maybe related, maybe not. I have power banks that will automatically turn off after 4-5 seconds if not enough power is being drawn. Can you check the power bank by plugging in a phone to see if it charges for more than 4-5 seconds.

santaimpersonator, 2021-09-30:
We ran into this issue recently. Our problem came from how the power control IC operated (on our updated batteries), which resulted in a USB port that was prioritized and always worked. However, the other ports browned out in the middle of the boot sequence. From what we could determine, this was probably because that prioritized port was being checked by the power control IC during that time. I made a post with a few more details in this discussion: https://github.com/NVIDIA-AI-IOT/jetbot/discussions/470#discussioncomment-1281752

closed 2022-07-18 by jaybdub