NVIDIA-AI-IOT/jetbot #364: road following error in Display Live Image code

Eljayce, 2021-01-15:
When executing the (latest) example notebook, road_following/data_collection.ipynb, the Display Live Image code below. 

```
camera = Camera()

widget_width = camera.width
widget_height = camera.height

image_widget = widgets.Image(format='jpeg', width=widget_width, height=widget_height)
target_widget = widgets.Image(format='jpeg', width=widget_width, height=widget_height)

x_slider = widgets.FloatSlider(min=-1.0, max=1.0, step=0.001, description='x')
y_slider = widgets.FloatSlider(min=-1.0, max=1.0, step=0.001, description='y')

def display_xy(camera_image):
    image = np.copy(camera_image)
    x = x_slider.value
    y = y_slider.value
    x = int(x * widget_width / 2 + widget_width / 2)
    y = int(y * widget_height / 2 + widget_height / 2)
    image = cv2.circle(image, (x, y), 8, (0, 255, 0), 3)
    image = cv2.circle(image, (widget_width / 2, widget_height), 8, (0, 0,255), 3)
    image = cv2.line(image, (x,y), (widget_width / 2, widget_height), (255,0,0), 3)
    jpeg_image = bgr8_to_jpeg(image)
    return jpeg_image

time.sleep(1)
traitlets.dlink((camera, 'value'), (image_widget, 'value'), transform=bgr8_to_jpeg)
traitlets.dlink((camera, 'value'), (target_widget, 'value'), transform=display_xy)

display(widgets.HBox([image_widget, target_widget]), x_slider, y_slider)
```

results in this error:

```
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jetbot-0.4.1-py3.6.egg/jetbot/camera/opencv_gst_camera.py", line 45, in _capture_frames
    self.value = image
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 588, in __set__
    self.set(obj, value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 577, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1210, in _notify_trait
    type='change',
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1215, in notify_change
    return self._notify_observers(change)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1252, in _notify_observers
    c(event)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 361, in _update
    self._transform(change.new))
  File "<ipython-input-2-8bd7c62a3894>", line 19, in display_xy
    image = cv2.circle(image, (widget_width / 2, widget_height), 8, (0, 0,255), 3)
TypeError: integer argument expected, got float

TypeErrorTraceback (most recent call last)
<ipython-input-2-8bd7c62a3894> in <module>
     24 time.sleep(1)
     25 traitlets.dlink((camera, 'value'), (image_widget, 'value'), transform=bgr8_to_jpeg)
---> 26 traitlets.dlink((camera, 'value'), (target_widget, 'value'), transform=display_xy)
     27 
     28 display(widgets.HBox([image_widget, target_widget]), x_slider, y_slider)

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in __init__(self, source, target, transform)
    337         _validate_link(source, target)
    338         self.source, self.target = source, target
--> 339         self.link()
    340 
    341     def link(self):

/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py in link(self)
    342         try:
    343             setattr(self.target[0], self.target[1],
--> 344                     self._transform(getattr(self.source[0], self.source[1])))
    345         finally:
    346             self.source[0].observe(self._update, names=self.source[1])

<ipython-input-2-8bd7c62a3894> in display_xy(camera_image)
     17     y = int(y * widget_height / 2 + widget_height / 2)
     18     image = cv2.circle(image, (x, y), 8, (0, 255, 0), 3)
---> 19     image = cv2.circle(image, (widget_width / 2, widget_height), 8, (0, 0,255), 3)
     20     image = cv2.line(image, (x,y), (widget_width / 2, widget_height), (255,0,0), 3)
     21     jpeg_image = bgr8_to_jpeg(image)

TypeError: integer argument expected, got float
```

Any ideas how to fix this?

DingoOz, 2021-01-16:
Just to check, have you tried restarting the kernels in Jupyter to see if that fixes it? Or even, a reboot of the Jetson to make sure there are not previous commands to the Camera causing issues.

Eljayce, 2021-01-18:
I have tried restarting the kernels in Jupyter and reboot of the Jetson, both didn't work

TennesseePilot, 2021-02-08:
I experienced the same problem as above. The jetbot setup was installed about 4 weeks ago. I have run the classification example program and had good results with it. I was able to gather data, train and run the Jetbot on a tabletop so I think I have everything setup correctly. Thanks for any help that anyone can provide...

TennesseePilot, 2021-02-08:
I think I have found the problem. These 2  lines are expecting an integer and are getting a float because the screen coordinates are divided by 2.
=======
--> 19     image = cv2.circle(image, (**widget_width / 2,** widget_height), 8, (0, 0,255), 3)
     20     image = cv2.line(image, (x,y), (**widget_width / 2**, widget_height), (255,0,0), 3)
=======
So I changed the code slightly in these two lines so the division by 2 produces an integer instead of a float.
=======
 image = cv2.circle(image, (int(widget_width / 2), widget_height), 8, (0, 0,255), 3)
    image = cv2.line(image, (x,y), (int(widget_width / 2), widget_height), (255,0,0), 3)
=======

I haven't completed the notebook yet but I have gone thru it and do not get an error anymore.

Hope this will help.....

TennesseePilot, 2021-02-08:
I think I have found the problem. These 2 lines are expecting an integer and are getting a float because the screen coordinates are divided by 2.
I changed the 2 lines

TennesseePilot, 2021-02-08:
I think I have found the problem. These 2 lines are expecting an integer and are getting a float. I modified these 2 lines to change the float to integer:
*************
 ->19     image = cv2.circle(image, (widget_width / 2, widget_height), 8, (0, 0,255), 3)
 ->20     image = cv2.line(image, (x,y), (widget_width / 2, widget_height), (255,0,0), 3)
*************
to this:
*************
image = cv2.circle(image, (int(widget_width / 2), widget_height), 8, (0, 0,255), 3)
    image = cv2.line(image, (x,y), (int(widget_width / 2), widget_height), (255,0,0), 3)
*************

This got rid of all errors. I haven't gone thru the entire notebook yet. If I run into another problem I'll report back.

closed 2022-07-15 by jaybdub