NVIDIA-AI-IOT/jetbot #368: jetbot 2gB camera crash

jaybdub, 2021-01-15:
ZmqCamera  should help with the second issue (not closing camera at end of notebook).  It actually may be possible to connect multiple notebooks simultaneously using ZmqCamera (not that I recommend this).

The camera crashing on 2GB likely has to do with the argus camera daemon running out of memory.  Both camera interfaces depend on the argus daemon, so it likely effects both.  To clarify, the issue I understand is that if the argus daemon runs out of physical RAM to use, it can crash.  

The way we prevented this with JetBot was by limiting the memory used by other processes (like the Jupyter Lab / Python).  This helps ensure that the argus daemon has enough physical RAM available to use.  Other processors will use SWAP memory when they hit their memory limit, so they still work.

Perhaps the memory limits aren't set on your jetson nano 2GB for some reason?

Do you mind sharing the output of

```bash
sudo docker stats
```

This should show the memory limit set for each docker container.

Best,
John

_Originally posted by @jaybdub in https://github.com/NVIDIA-AI-IOT/jetbot/issues/359#issuecomment-761202580_

bezineb5, 2021-01-16:
Initially I didn't put memory limits - it was not obvious why I would want to do that.

This is the docker stats before running the collision detection notebook:
```
CONTAINER ID        NAME                CPU %               MEM USAGE / LIMIT     MEM %               NET I/O             BLOCK I/O           PIDS
8ac7010fce0a        jetbot_jupyter      0.00%               63MiB / 1.933GiB      3.18%               0B / 0B             94.4MB / 45.1kB     4
0ace7336a697        jetbot_display      1.89%               218.8MiB / 1.933GiB   11.05%              0B / 0B             679MB / 0B          2
1ccd7bcca6d6        jetbot_camera       6.97%               36.57MiB / 1.933GiB   1.85%               0B / 0B             31.4MB / 1.35MB     14
```

And the stats after loading the model:
```
CONTAINER ID        NAME                CPU %               MEM USAGE / LIMIT     MEM %               NET I/O             BLOCK I/O           PIDS
8ac7010fce0a        jetbot_jupyter      0.04%               970.9MiB / 1.933GiB   49.04%              0B / 0B             2.65GB / 105MB      19
0ace7336a697        jetbot_display      0.00%               8.641MiB / 1.933GiB   0.44%               0B / 0B             810MB / 19.4MB      2
1ccd7bcca6d6        jetbot_camera       0.00%               2.848MiB / 1.933GiB   0.14%               0B / 0B             33.8MB / 1.99MB     14
```

Then I used the values suggested in the [docker containers readme](https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker) and it became much better:
```
CONTAINER ID        NAME                CPU %               MEM USAGE / LIMIT     MEM %               NET I/O             BLOCK I/O           PIDS
239b3db38213        jetbot_jupyter      31.51%              379.1MiB / 500MiB     75.81%              0B / 0B             3.08GB / 1.2GB      22
4abfb8cf3b37        jetbot_display      0.00%               204.7MiB / 1.933GiB   10.34%              0B / 0B             634MB / 0B          2
d48dbc84b95e        jetbot_camera       8.77%               35.29MiB / 1.933GiB   1.78%               0B / 0B             154MB / 1.35MB      14
```

And it solved the issue with the camera!
My suggestion would be to rephrase the readme with something like: "It the jetson runs out of memory, it may kill the camera process. If you experience camera instability, set memory limits on the Jupyter container:
export JETBOT_JUPYTER_MEMORY=500m
export JETBOT_JUPYTER_MEMORY_SWAP=3G
On the Jetson Nano 2Gb, this is required to run the examples.
"
(obviously, feel free to improve/correct the inexactitudes!)

Thanks a lot for your help!

jaybdub, 2021-01-16:
Fantastic!  

I think the phrasing you suggested sounds great.  I'll work to incorporate that soon.

Please reach out if you have any other issues or feedback, it's much appreciated :) 

Best,
John

jaybdub, 2021-01-17:
Actually, there was an issue with the docker configuration such that the memory limits weren't being set correctly.

Should be fixed in this PR

https://github.com/NVIDIA-AI-IOT/jetbot/pull/370

So that 2GB will set 500MB by default.  We may still want to explain the behavior / reasoning as you suggested.

Best,
John

bezineb5, 2021-01-17:
Indeed, good catch!
Thanks for this.

niiita, 2022-01-28:
Hi I'm also having the same problem that whenever I try accessing the camera my nano crashes.
I'm using the recommended power sources and tried defaulting it to 5V.
So far, I've followed the readme @ https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker
I tried to follow this issue but when I try to get the memory limits I just get a loop of the headers
[image]

Any help would be appreciated.