NVIDIA-AI-IOT/jetbot #375: Release with JetPack 4.5?

oschulz, 2021-01-24:
Quick question, are there plans for a release with JetPack 4.5 in the near future?

DingoOz, 2021-01-26:
I haven't tried it with 4.5 yet. Have you run into any issues with JetPack 4.5, can you detail them here?

oschulz, 2021-01-26:
Not issues, really, I just needed the a new CUDA version (JetPack 4.5 have CUDA 10.2). However, was able to make things work using JetPack 4.5 directly now, so there's no urgency.

tcorreia2019, 2021-01-31:
I am following the instructions in https://jetbot.org/master/software_setup/docker.html.  I had previously done steps 1&2 when got the jetson nano 2gb a week ago and installed Jetpack 4.5.  Ran steps 3 and 4 without a problem.  But when I run Step 5 I get the output below
+++++++++
abc@abc-desktop:~/jetbot/docker$ ./enable.sh $HOME
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
docker: invalid reference format.
See 'docker run --help'.
docker: invalid reference format.
See 'docker run --help'.
++++++++++
In looking at the $HOME/jetbot/docker/configure.sh I see checks for $L4T_VERSION = 32.4.3 and 32.4.4 but no check for the version 32.5.0 which is what you get with JetPack 4.5

tokk-nv, 2021-02-03:
Hi all,

We just made a release of v0.4.3 of jetbot that supports JetPack 4.5.
https://github.com/NVIDIA-AI-IOT/jetbot/releases/

The SD card images are available for Jetson Nano 2GB as well as for the original 4GB Jetson Nano Developer Kit.
http://jetbot.org/master/software_setup/sd_card.html

Please let us know if you run into any issue or find any unmet needs related to JetPack 4.5.

tcorreia2019, 2021-02-03:
Awesome.  Thanks for the update

oschulz, 2021-02-03:
Thanks a lot!

robbuckley, 2021-02-20:
hi, the pre-trained  model for the `object_following` example doesnt seem to work. I tried the file linked here for "JetBot SD Card Version" 0.4 https://github.com/NVIDIA-AI-IOT/jetbot/blob/master/docs/examples/object_following.md. That page warns that it only works with JetPack 4.3..

Any chance of a model compatible with 4.5? Or some tips on how to convert it, if thats possible? 

many thanks

jaybdub, 2021-03-20:
Hi @robbuckley ,

Thanks for reaching out!

Unfortunately, we haven't yet added support to the object detection model for JetPack 4.5.

This object detector consistently causes errors with new versions of software, likely due to the nuances in the post-processing.

We hope to develop a simple alternative to work past this.  

For the time being, you may be able to look at 

https://github.com/qfgaohao/pytorch-ssd

Or a similar open-source repository.  I'm happy to help how I can getting this to work for you.  I believe this project 

https://github.com/NVIDIA-AI-IOT/ros2_torch_trt/tree/master/trt_live_detector/trt_live_detector

Has success converting this model to TensorRT.

Please let me know if this helps or you run into any issues.

Best,
John

OZtwo, 2021-04-29:
> Hi all,
>
> We just made a release of v0.4.3 of jetbot that supports JetPack 4.5.
> https://github.com/NVIDIA-AI-IOT/jetbot/releases/
>
> The SD card images are available for Jetson Nano 2GB as well as for [...]

Yes, but also you stated you must first install this then install JetPack 4.5 using SDK Manager? I have this installed yet there is no Jetpack on it?

You have a notice on Jetbot.org:

**Attention
To use a sdcard image that is based on JetPack 4.5, you need to run through the initial ("oem-config") setup using the original JetPack 4.5 Jetson Nano SD card image (Nano 2GB, Nano (4GB)) first. (two links given)
_This step will configure the bootloader on Jetson Nano hardware as needed for the latest SD card we released for JetBot._**

fifise77, 2021-09-07:
Hello,

did you add support to the object detection model for JetPack 4.5. yet ?

I cannot find a pre trained model ssd_mobilenet_v2_coco.engine that work right now

OZtwo, 2021-09-07:
Not as of yet. I took a break for the robot and focused more on my new 3d printer to then go back to the robot itself when I feel good at using FreeCAD.

ArrowRaider, 2021-09-08:
JetPack 4.6 is out now