NVIDIA-AI-IOT/jetbot #383: Connection Issue- wlan not showing up

spy-maker, 2021-01-26:
I am facing connectivity issue in my jetbot. I already setup start up service and jupyter service but it does not display wlan address when I start up jetbot. It only shows the address when I connect jetbot with computer using hdmi cable. Does anyone know what is the cause of it and how to fix it ?

DingoOz, 2021-01-26:
Can you detail your issue further?

Do you mean that the OLED screen on the robot is not showing the IPv4 address? But that you can find out the address if you plug in a screen and keyboard and go to a terminal and run "ipconfig" or similar?

Further information that is good to know. Presently, when you do find out the address are you able to access it from a computer or does it still block you somehow?

Additional information that would be good. Is your Jetbot connecting to a router? Is you PC connecting to the same router?
Can your router see the Jetbot if you log into an admin console?

Have you added the SSID and password of your wifi network to your Jetbot and confirmed it worked at all?

spy-maker, 2021-01-26:
> Can you detail your issue further?
>
> Do you mean that the OLED screen on the robot is not showing the IPv4 address? But that you can find out the address if you plug in a screen and keyboard and go [...]

I have done everything and all working good but the only problem is that the jetbot OLED screen does not shows wlan address it just shows as none. But when I connect the jetbot with pc using hdmi cable it shows the wlan address. It shows the ip when i enter ifconfig. So that is fine. Both the pc and jetbot is in same wifi network. No problem with it. 
The main concern is that the jetbot doesnt show the wlan address when startup but it shows after I connect with pc using hdmi cable. There is no issue with connection. But just this part.

DingoOz, 2021-01-26:
If the OLED screen is not showing the IP Address then somehow this line in stats.py is not returning the correct information (or not being called):
https://github.com/NVIDIA-AI-IOT/jetbot/blob/7b371ebf1f2dc3a5b25309cddc87928ee628807c/jetbot/apps/stats.py#L89

If you haven't done so recently, try updating to the latest version of the JetBot code here.

...

As a workaround, if this lack of IP Address is causing an issue, you could try setting a static IP for your JetBot. The easiest way to do this is to change the settings on your router to allocate the same IP address to the MAC address of your JetBot's Wifi card.

This way you don't need to rely on the OLED screen, the IP Address will always be what you have set it to.

spy-maker, 2021-01-26:
> If the OLED screen is not showing the IP Address then somehow this line in stats.py is not returning the correct information (or not being called): [...]

Thanks for the reply mate. I have already done static ip address settings. I am currently working on few jetpack images but the first one only got this issue the rest all dont have.

closed 2021-02-16 by spy-maker