NVIDIA-AI-IOT/jetbot #384: Do I need to flash an SD card to get jetbot to work? (i.e. why isn't there an sh file for this type of stuff)

josh-github1, 2021-01-30:
question in the title ^

josh-github1, 2021-01-30:
I'm trying to run jetbot on the original nano with Jetpack 4.3

DingoOz, 2021-01-30:
I am not really sure but I assume you can clone this repo onto your existing Jetpack 4.3 and run the docker (see info about the docker here: https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker)

Have you tried something like this already and get an error at any stage?

jaybdub, 2021-02-04:
Hi @josh-github1,

Thanks for reaching out.  We've considered maintaining a native configuration script, but for the time being we've chosen to maintain just a docker image (and SD card images which wrap this) for JetBot.  This largely has to do with maintenance effort.

That said, we attempted something like this in the past with [JetCard](https://github.com/NVIDIA-AI-IOT/jetcard/blob/master/install.sh).

You might be able to modify this for your purposes.  It includes most of the requirements for JetBot, but will need some tweaking.

Please let me know if this helps or you run into issues.

Best,
John

josh-github1, 2021-02-09:
@DingoOZ I'll try this in a bit.. I was spending some time trying to get a depth camera to work on the jetson nano

culebracut, 2021-02-11:
I'm also interested in trying this.  For example the remote control aspects of jetbot don't have pytorch or tensorflow as dependencies so maybe an easy starting point.

josh-github1, 2021-02-24:
I'm working on this now.. seems like the 4.2.2 branch for the jetcard repo sort of works but there's a lot of errors

josh-github1, 2021-02-24:
> > I ran into the following GPIO error (could be a path issue or missing config files) in the install.sh script:
> > $sudo -S cp /opt/nvidia/jetson-gpio/etc/99-gpio.rules /etc/udev/rules.d/
> > [sudo] password for kaisark:
> > cp: cannot stat '/opt/nvidia/jetson-gpio/etc/99-gpio.rules': No such file or directory
> > ~$ ll /opt/nvidia/jetson-gpio
> > ls: cannot access '/opt/nvidia/jetson-gpio': No such file or directory
> > $ ll /etc/udev/rules.d
> > total 104
> > drwxr-xr-x 2 root root 4096 Sep 14 12:32 ./
> > drwxr-xr-x 4 root root 4096 Sep 16 12:53 ../
> > -rw-r--r-- 1 root root 58549 Sep 14 12:32 70-snap.core.rules
> > -rw-r--r-- 1 root root 616 Jun 26 00:36 90-alsa-asound-tegra.rules
> > -rw-r--r-- 1 root root 175 Jun 26 00:36 91-xorg-conf-tegra.rules
> > -rw-r--r-- 1 root root 962 Jun 26 00:36 92-hdmi-audio-tegra.rules
> > -rw-r--r-- 1 root root 208 Jun 26 00:36 99-nv-l4t-usb-device-mode.rules
> > -rw-r--r-- 1 root root 1137 Jun 26 00:36 99-nv-l4t-usb-host-config.rules
> > -rw-r--r-- 1 root root 425 Jun 26 00:36 99-nv-ufs-mount.rules
> > -rw-r--r-- 1 root root 634 Jun 26 00:36 99-nv-wifibt.rules
> > -rw-r--r-- 1 root root 1834 Jun 26 00:36 99-tegra-devices.rules
> > -rw-r--r-- 1 root root 130 Jun 26 00:36 99-tegra-mmc-ra.rules
> 
> You shouldn't have to do this because of this - https://forums.developer.nvidia.com/t/there-is-no-99-gpio-rules-file-in-the-latest-jetson-gpio-library/118619

josh-github1, 2021-02-24:
I skipped installing pytorch, tensorflow, and traitlets because these were installed in the past.

josh-github1, 2021-02-24:
> I am not really sure but I assume you can clone this repo onto your existing Jetpack 4.3 and run the docker (see info about the docker here: https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/docker)
> 
> Have you tried something like this already and get an error at any stage?

I haven't tried docker yet and as of now- currently attempting to build everything from source though I am unsure if this is a good idea

josh-github1, 2021-02-24:
For this command to work from the install.sh file for jetcard - 

# install jupyter lab
echo $password | sudo -S apt install -y nodejs npm

These commands need to be run:
sudo -s apt install -y libssl1.0-dev
sudo -s apt install -y nodejs-dev
sudo -s apt install -y node-gyp

josh-github1, 2021-02-24:
Then nodejs had to be upgraded via https://askubuntu.com/questions/426750/how-can-i-update-my-nodejs-to-the-latest-version

to run this command

echo $password | sudo -S jupyter labextension install @jupyter-widgets/jupyterlab-manager

closed 2022-07-15 by jaybdub