NVIDIA-AI-IOT/jetbot #385: I flashed last released Jetson Nano 2GB	4.5 v0.4.3, but it does'n boot.  The led turns on but nothing happens on display

miguelisa, 2021-02-04:


jaybdub, 2021-02-04:
Hi miguelisa,

Thanks for reaching out!  

Have you used a JetPack 4.5 SD card image before?  Or are you coming straight from an older JetPack version?  

Best,
John

miguelisa, 2021-02-04:
Hi jaybdub

I have used a formated (blank) SD Card to flash jetbot-043_nano-2gb-jp45.zip image, and them I tried to boot jetson nano with it

jaybdub, 2021-02-05:
I see, thanks for sharing.  We may have seen something similar when we first developed the jetbot-043_nano-2gb-jp45.zip image, but we weren't able to reproduce it in testing.  One difference is that we had previously used a JetPack 4.5 image, which we think may have had an impact (at this point, we're not sure).

If it is not an inconvenience, could you try flashing the standard JetPack 4.5 Image https://developer.nvidia.com/EMBEDDED/jetpack and running through the standard (non JetBot) setup on your Jetson?  It may be good to verify this is working as intended.

Best,
John

miguelisa, 2021-02-05:
Ok, thanks jaybdub

Regards
Miguel

mmatesic01, 2021-02-08:
Hi,
Looks like It is something wrong with the image file. Jetbot does not boot with it.
I have downloaded the file twice withe the same result. (used Etcher to upload IMG to SD card). Could be corupted IMG file?

jaybdub, 2021-02-08:
Hi mmatesic01,

Did you try flashing the regular Jetson Nano image here

https://developer.nvidia.com/EMBEDDED/jetpack

And run through the ubuntu setup with a keyboard and monitor attached to your Jetson Nano?  

Currently, I think it may have to do with a bootloader change introduced in JP4.5 that may require running through the original Jetson Nano image setup.  After you have flashed with that Image, it is possible that the JetBot image would work afterwords.

Apologies for any frustration this causes.  Please let me know if you have any questions.

Best,
John

mmatesic01, 2021-02-09:
HI jaybdub,
I'll try that and report back.

mmatesic01, 2021-02-09:
Hi jaybdub,
I can confirm that your approach worked.
Thank you.

jaybdub, 2021-02-09:
@mmatesic01 Thanks for following up with this!  Very helpful to know.

Sorry for any frustration this caused.  Hopefully others will benefit from your experience.

Best,
John

Abdulsattar21, 2022-02-28:
@jaybdub 
Hello,
I'm trying to boot the jetson nano (4gb version), but each time I get the same result where a green light starts working but nothing happing, also I followed the instructions that you've mentioned before and download the latest version of jetpack but also got the same result, is there anything I can do to make the jetson nano work?