NVIDIA-AI-IOT/jetbot #390: Jetson Nano doesn't boot with new image

w1ngedshadow, 2021-02-14:
Hello, I flashed new release of pre-built JetBot SD card image for Jetson Nano 4GB (jetbot-043_nano-4gb-jp45.zip) and it doesn't boot. I read issue #385 and tried to boot with original nvidia image but it doesn't boot again and I have no Nvidia logo displayed. After that none of my different versions images don't boot . Before this incident my Jetson Nano had worked correctly. How to back my Jetson Nano to work?

jaybdub, 2021-02-16:
hi @w1ngedshadow ,

Thanks for reaching out!

I believe this is likely related to the issue below

https://github.com/NVIDIA-AI-IOT/jetbot/issues/385

The solution is to run through the SD card setup for the original JetPack 4.5 Jetson Nano SD card image. [here](https://developer.nvidia.com/embedded/jetpack).  This step will configure the bootloader as needed for the latest SD card we released for JetBot.

Please let me know if this helps or you run into any issues.

Best,
John

w1ngedshadow, 2021-02-21:
> hi @w1ngedshadow ,
>
> Thanks for reaching out!
>
> I believe this is likely related to the issue below
>
> #385
>
> The solution is to run through the SD card setup for the original JetPack 4.5 Jetson Nano SD [...]

Thanks, but I was unable to restore the boards as you advised. I used NVIDIA SDK Manager and force recovery mode on jetson ant it works for me. Now all is correct.

OZtwo, 2021-04-29:
> > hi @w1ngedshadow ,
> > Thanks for reaching out!
> > I believe this is likely related to the issue below
> > #385
> > The solution is to run through the SD card setup for the original JetPack 4.5 Jetson Nano SD card image. [here](https://developer.nvidia.com/embedded/jetpack). This step will configure the bootloader as needed for the latest SD card we released for JetBot.
> > Please let me know if this helps or you run into any issues.
> > Best,
> > John
> 
> Thanks, but I was unable to restore the boards as you advised. I used NVIDIA SDK Manager and force recovery mode on jetson so it works for me. Now all is correct.

They have posted this and it makes no sense as you need to use SDK Manager and a second Linux system to install JetPack 4.5?

**Attention
To use a sdcard image that is based on JetPack 4.5, you need to run through the initial ("oem-config") setup using the original JetPack 4.5 Jetson Nano SD card image (Nano 2GB, Nano (4GB)) first. (two links given)
This step will configure the bootloader on Jetson Nano hardware as needed for the latest SD card we released for JetBot.**

jaybdub, 2021-04-29:
Hi @OZtwo,

Thanks for sharing your experience, and apologies for any confusion.

You should only need to download the JetPack 4.5 SD Card image directly and flash it to the SD card, without needing to interface with the SDK manager (so no second linux machine should be needed).  After that, when you boot Jetson for the first time, it requires you to run through the steps of setting your Linux username, time zone, etc.  I believe during this process the Bootloader is updated for an update specific to JetPack 4.5+.  Since the JetBot SD card image is already pre-configured, it wont perform this step on your machine, so if its your first time using Jetson with an SD card based on JetPack 4.5, you might need to run through this step separately.

Did you try flashing the plain JetPack 4.5 SD card image, running through the setup, and then switching back to the JetBot SD card image?  

If you did this and it didn't work, I'll need to dig deeper to figure out what's going on.

If you haven't yet tried this, I would give it a try.  On our end, we'll think about how we can clarify the wording so that it's less confusing.  If you have suggestions on what would help here, it's greatly appreciated.

Please let me know if this helps or you have any questions / run into issues.

Best,
John

OZtwo, 2021-04-29:
Thanks for you fast reply. I have already done this yet after installing (two times) ended up at a blank login screen  and an error message printed every 10 second. It was then I went back to Jetbot.org and found this odd message:

**Attention
To use a sdcard image that is based on JetPack 4.5, you need to run through the initial ("oem-config") setup using the original JetPack 4.5 Jetson Nano SD card image (Nano 2GB, Nano (4GB)) first. (two links given)
This step will configure the bootloader on Jetson Nano hardware as needed for the latest SD card we released for JetBot.**

In the above message was given a link to this file: jetson-nano-jp451-sd-card-image.zip

When installing this base file all works great with full GUI yet no Jetpack which I hadn't installed yet. When looking into ways to get the second image installed I was pointed to the need to use SDK manager and install the second img using a host computer.

As well, I have seen four other posts here with people asking about the same question as I with others not sure as to why they are at a blank screen with with the same error message repeating every 10 seconds about no Ic2 (or something) not found.

At the end you can see they know of the issue and stated how to fix it yet I'm without a second Linux box. :(

Why are these two images not merged into one?

OZtwo, 2021-04-29:
Update: I went back one version to:

**Jetson | Nano (4GB) | 4.4.1 | 0.4.2 | jetbot-042_nano-4gb-jp441.zip**

With this version I have the exact same issue. I'm using a Jetson Nano B01 version.

At boot the system will log into **Ubuntu 18.04.5 TS nano-4GB-jp441 tty1** full screen text mode login.
As well I still get the error: **tegra-i2c 70000c400.i2c: no acknowledge from address 0x3c**

This message will show up about ever 10~ seconds. After login I'm stuck in text mode with the error scrolling down the screen. The message will keep showing up even as it's shutting down. On my last message I posted a link to another user who has the exact same issue with a screen shot.

zlite, 2021-04-29:
Do you actually have an OLED screen connected? If not, that would explain this.

OZtwo, 2021-04-29:
interesting. Yes when I got the error for the very first time unplugged it and then just to test again plugged it back in. BUT I'm using the Sparkfun jetbot kit so yes the this version of the monitor my not work. but still stuck at the login screen and did find the 'omg-config' yet again it seems I need to run this on a second PC? Of course all I want is it to boot into the GUI so I can THEN remove the GUI. :) But at the end it seems this is a known issue since on jetbot.org site they have a warning message that the new image will not work unless you install the older image first and then use skd manager? The site says you need to first flash (jetson-nano-jp451-sd-card-image.zip) before moving on to the next step flashing of the new images?!?

OZtwo, 2021-04-29:
fyi the image they say you need to download first seem to only be the OS itself with the GUI.

zlite, 2021-04-29:
Maybe the Sparkfun OLED has a different I2C address?

OZtwo, 2021-04-29:
> Maybe the Sparkfun OLED has a different I2C address?

yeah my guess. But at the end still all around learning this new world. :)

OZtwo, 2021-04-29:
> > Maybe the Sparkfun OLED has a different I2C address?
> 
> yeah my guess. But at the end still all around learning this new world. :)

my very first unknown guess was that it had a issue with the HAT (if that is right) so I unplugged the 40pin QWIIC thing. I wouldn't be at such a loss (ok maybe I would) if not for them stating you needed to install two images sounding as if they knew you would run into this issue if you didn't install the first older image and then use a second PC to install the newer images? And of course I'm stuck at why even have such a warning and not give us a full images with both? (which again I kind of understand since if they did give a full version you then would need to flash all data and start from the beginning. This way you can keep all data and update it to the latest version.) BUT then again we have so many here without this issue who are having a GREAT time programming their bots learning new things. :)

santaimpersonator, 2021-06-16:
There is a service that runs in the background to display the IP address on the OLED screen. The SparkFun kit uses a different OLED screen with a different I2C address... hence the error message from the service. You can easily disable or stop the service so that the error message doesn't pop up.

[SparkFun provides an image for their hardware and instructions for setting up the jetbot software for their hardware](https://learn.sparkfun.com/tutorials/assembly-guide-for-sparkfun-jetbot-ai-kit-v20/4-software-setup-guide-from-nvidia).

*\***Note:** I am not sure how the restructuring of the QSPI flash from Jetpack 4.5 will affect the compatibility with older images utilizing previous Jetpack versions.*

BharatDadwaria, 2021-06-29:
Hello,
I have booted Jetpack 4.5.1 in SD card and started booting up the jetson nano for the first time, but it didn't show any activity. However the green light is on but nothing happening and just showing the blank screen. What may be the possible upcoming actions?

santaimpersonator, 2021-06-29:
@BharatDadwaria Please see the [second comment](https://github.com/NVIDIA-AI-IOT/jetbot/issues/390#issuecomment-780076567) or the call out at the bottom of **Step 1** here: http://jetbot.org/master/software_setup/sd_card.html

BharatDadwaria, 2021-07-03:
@santaimpersonator  I did follow every single step described above and by you but still the jetson nano is not booting up. The green light is on but nothing happening on screen. The condition of the jetson nano is also brand new.

santaimpersonator, 2021-07-03:
Maybe you have a bad board 🤷. I'd try the steps below:
1. Make sure you are flashing the image from the downloaded [`jetson-nano-2gb-jp451-sd-card-image.zip`](https://developer.nvidia.com/jetson-nano-2gb-sd-card-image) (2GB) or [`jetson-nano-jp451-sd-card-image.zip`](https://developer.nvidia.com/jetson-nano-sd-card-image) (4GB) files for your board
    - I recommend using the linked images to avoid issues. Don't use a custom image or one made by the SDK.
2. Make sure you are using the image for your board's memory size.
    - I don't think the 2GB and 4GB images are cross compatible.
3. Make sure your display is working and set to the correct input.
    - It's always good to test with another device (with the same cable) to make sure it isn't a bad cable.
4. Make sure the compute module is fully inserted in carrier board.
    - Probably not a problem, but it doesn't hurt to double check and re-seat the module.
5. Make sure the SD card is locked in (it should be pretty much flush with the edge)
6. Make sure you are powering the board properly
    - The 2GB board needs to be powered through the USB-C connector
    - The 4GB board get powered through the USB micro-B connector, unless you use close the jumper to power the board through the barrel jack
    - Make sure if you are powering the board through the USB connector, that the power supply can source enough current (3A) so that the board isn't browning out

BharatDadwaria, 2021-07-05:
@santaimpersonator  I have followed and checked all the steps given above. I did try using the [jetpack 4.5.1](https://developer.nvidia.com/jetson-nano-sd-card-image), version 4.5 ,jetbot image but nothing actually working here. I have checked the display which is working fine. The condition of the jetson nano is brand new (i ordered it on 13th June), I have ordered another SD card(64GB) and formatted it as exFAT.  Now I am very exhausted troubleshooting the jetson nano. Is there any other step which may work in this case. I am mentioning my steps if i am missing something please let me know.

- Booted jetpack 4.5.1 on 16GB SD card -> green light is on and nothing happening on the screen.
- Booted some other older version from archive -> green light is up and nothing happening on the screen
- ordered brand new USB cable (2.5A) and new SD card(64GB) 
- Booted again with [jetbot](: http://jetbot.org/master/software_setup/sd_card.html) as described by you on new SD card (formatted as exFAt). -> Same issue
- Tried jetpack 4.5.1 and older jetpack version on new SD card -> green light is up and nothing happened

santaimpersonator, 2021-07-06:
I should also note on the display, don't use an older display where you need an adapter. I once, had an issue with a VGA display that used a HDMI to VGA adapter. Once, I switched to a more modern monitor with an HDMI input jack (also worked with a TV with an HDMI jack)... the GUI Desktop showed up.

On your SD card:
- I've never used a 16GB card, just because they don't have enough space to create a swapfile... which means they run like slugs.
- You'll probably want at least a 32GB card anyways to run the examples, since with the pre-trained models you need to download are fairly large.
- Make sure to utilize a name brand card (like Kingston, SanDisk, PNY, etc.) that is at least class 10 (you get what you pay for).
- With a cheap card, it might not meet the required specs and will likely just cause more issues.

Follow up questions:
- Do you have the 2GB or 4GB board?
- Which exact image are you using?
- When you say nothing is happening on the screen is is blank (black) or do you have a splash screen with the Nvidia logo?
    - If black, does your monitor tell you if it detects an input signal (vs say if the HDMI cable were unplugged)?

Again, the issue could be the board. You will probably find more resources on the [Nvidia forum](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70).

skribtsov, 2021-08-03:
Hello, 

Is it possible to create SD card that would start working immediately on a fresh/new/unitialized Nvidia-nano board? I did read this **"To use a sdcard image that is based on JetPack 4.5, you need to run through the initial ("oem-config") setup using the original JetPack 4.5 Jetson Nano SD card image (Nano 2GB, Nano (4GB)) first."** And this approach does work. But we need to initialize more than 100 devices and initiation EACH device with the official image first is not a variant. What should we do so that our SD card image would immediately start working on a new (un-initialized) Nvidia-nano. Can we make our image so that it would just as the official one do this, cite : ".. configure the bootloader on Jetson Nano hardware as needed for the latest SD card we released for JetBot".

Hope explained the question well.. I would really appreciate a working advice. 
Thank you, 
Pavel

santaimpersonator, 2021-08-03:
@skribtsov That really should be posted under a new topic; however, check out the [documentation here](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/flashing.html#wwpID0E0EE0HA).

huccpp, 2021-11-26:
Dear all,
I try to build  jetbot recently and also have similar issue and still cannot get through.
I am using 4GB version B01.
I also read  #385 like others, boot the machines with an SD card flashed with jetson-nano-jp451-sd-card-image.zip
(also jp45 instead of 451 in another attempt), went through username password filling procedure and 
got the the nvidia desktop page, turn off and boot with the SD card flashed with the jetbot image jetbot-043_nano-4gb-jp45.zip.
I got a dark screen ('dark' means I could tell the machine is connected to the screen but it was black, 
not a complete "idle" that the signal was not detected, and I use display port cable.)
I also tried to download and flash again the the jetbot image and re-tested to make sure it was not any corruption during download or flashing.
I use jack to power the machine, with the jumper which disables the micro usb side.
I am new to jetson and I am grateful for any advice.
Thanks in advance.
Gucp

farzi2007, 2022-03-09:
Hello guys!

I'm stuck with the same issue as #385.
- Flashed 64GB Sd card with jetson-nano-jp-461 
- inserted the card to jetson nano 4GB
- Connected HDMI cable
- connected J48 jumper 
- powered using barrel jack 5V 4A Adapter.
- Result - Black Screen. 

- Performed flashing the image a few times.
- swapped SD cards
- Tried to power up using Micro USB cable after removing J48.

- All these cases , green LED light is stable 
- (feel like the board has rebooted three times - [the screen went to power saving mode and then turned on.]) 

-Should I try installing older version?
Thanks,
Farzeen

MMGomaa, 2022-05-25:
@farzi2007 did you solve it?

> Hello guys!
>
> I'm stuck with the same issue as #385.
>
> * Flashed 64GB Sd card with jetson-nano-jp-461
> * inserted the card to jetson nano 4GB
> * Connected HDMI cable
> * connected J48 jumper
> * powered [...]

did you find a solution for it? as i have same issue

FinTheHuman21, 2022-10-23:
I have the same problem with a jeston nano 2gb p3541. Good job Nvidia, last time I buy trash from you.

> @farzi2007 did you solve it?
>
>
> did you find a solution for it? as i have same issue

SjoerdTimovanVliet, 2023-05-25:
@MMGomaa @w1ngedshadow @huccup @BharatDadwaria Did any of you fix this problem? I have a similair issue and can't seem to solve it.

TonyJ-AI, 2024-07-15:
I resolved the myths. Check out details here: 

https://github.com/TonyJ-AI/Jetson-Nano-Dev-Kit-Image-Load-Error/tree/main