NVIDIA-AI-IOT/jetbot #394: Waveshare Jetbot PiOLED Display not working

abuelgasimsaadeldin, 2021-02-26:
Hi guys, 

Not sure if anyone had the same issue as I did but I bought 3 Jetbots from Waveshare and all 3 of the times when I finish the setup, the PiOLED display will turn on and start working (display the ip address along with some other information) but very shortly after it will stop working!

Everything works fine however in terms of the motor driver and the Jetbot moving and all but it is really quite troublesome connecting a monitor and mouse every time just to check the ip address if it had changed.

I'm really trying to pinpoint why this keeps happening and have come up with several assumptions and one of which could be the type batteries that I am using however I don't see anything wrong with it as it is recommended by Waveshare (i'm using 3.7V Panasonic batteries from the link below)

https://www.lazada.com.my/products/panasonic-ncr18650b-3400mah-37v-18650-rechargeable-battery-protected-button-top-flat-top-i1299600903-s3966140962.html? 

I would really appreciate if anyone could assist me in pinpointing why this keeps happening and if there is any solution which could bring back my OLED display to work again! Would really appreciate it thanks guys.

jaybdub, 2021-02-26:
Hi @abuelgasimsaadeldin ,

Thanks for reaching out!

Just to clarify, are you saying that the JetBot is shutting down unexpectedly, and then the IP address is re-assigned?

Best,
John

abuelgasimsaadeldin, 2021-02-26:
Hi @jaybdub,

Thank you for the prompt reply, no the Jetbot is working fine and does not shutdown the only issue is that nothing is getting displayed on PiOLED display upon boot, so the only way to obtain the IP address and connect to the Jetbot would be by connecting a monitor and mouse to the Jetbot and checking it manually.