NVIDIA-AI-IOT/jetbot #397: Object Detection using Jetson Nano 2GB

TonyBaston, 2021-03-01:
Is there a way to run Object Detection on a Jetson Nano 2GB? The docs say the Object Detection notebook only works with Jetpack 4.3, which does not run on the 2GB(from here: https://forums.developer.nvidia.com/t/object-detection-with-on-jetbot-using-jetson-2gb/169611), and I have confirmed that it does not run with 4.5 or 4.4 (I get the error ```AttributeError: ‘NoneType’ object has no attribute ‘create_execution_context’```).

ak-nv, 2021-03-01:
@TonyBaston I would suggest following method on Jetson Nano (4GB) and it works with real-time performance.

- Used [pytorch-ssd](https://github.com/qfgaohao/pytorch-ssd/blob/master/run_ssd_live_demo.py) github for pre-trained weights and inference script.
- Have a look at example integration using [torch2trt](https://github.com/NVIDIA-AI-IOT/torch2trt) here: ROS2 package for [Torch_TRT for object detection](https://github.com/NVIDIA-AI-IOT/ros2_torch_trt/blob/master/trt_live_detector/trt_live_detector/trt_detection_helper.py) 
- From above step, you can use [jetcam](https://github.com/NVIDIA-AI-IOT/jetcam) for CSI camera input instead of accepting input from [/Image Topic](https://github.com/NVIDIA-AI-IOT/ros2_torch_trt/blob/ea65bc045761cc7732fbb548369fa16219050e71/trt_live_detector/trt_live_detector/trt_detection_helper.py#L90)

Please let me know if you have any issues.

TonyBaston, 2021-03-04:
So I got pytorch-ssd, and got mobilenetv2-lite, and converted it to a trt model. Then in another notebook, I loaded the model like this:
```
from torch2trt import TRTModule
import torch
net = TRTModule()
net.load_state_dict(torch.load('mobilenetv2_trt.pth'))
```
I decided to use the pytorch-ssd predictor class as from a source code perspective it looks almost identical to the ros2 one. Though when I try to call it I get this as my output:
```
Inference time:  0.008100748062133789

RuntimeErrorTraceback (most recent call last)
<ipython-input-5-57f1bd129277> in <module>
      1 predictor=create_mobilenetv2_ssd_lite_predictor(net, candidate_size=200)
----> 2 predictor.predict(np.zeros((300,300,3)))

/workspace/jetbot/notebooks/working object detection/pytorch_ssd/vision/ssd/predictor.py in predict(self, image, top_k, prob_threshold)
     42             prob_threshold = self.filter_threshold
     43         # this version of nms is slower on GPU, so we move data to CPU.
---> 44         boxes = boxes.to(cpu_device)
     45         scores = scores.to(cpu_device)
     46         picked_box_probs = []

RuntimeError: CUDA error: unspecified launch failure
```
From more experimentation, it seems that when ```net.forward()``` is called, no error is thrown, yet only when I try to look at the outputs. For example if I called ```scores,boxes=net.forward(images)```, the above error(```RuntimeError: CUDA error: unspecified launch failure```) would be thrown if I tried to run ```scores[0][0]``` .
Also, I am running this on the Jetson 2GB, if that makes any difference.

tooyipjee, 2021-04-21:
@TonyBaston I am having the same issue. It seems like there really isn't any versions of JetBot that would work out of the box with the Nano 2GB. Got myself a 2GB as well and if I had known that I would have gotten the 4GB instead. 

Now I have a fully assembled JetBot Nano 2GB sitting in a corner doing nothing...

closed 2022-07-15 by jaybdub