NVIDIA-AI-IOT/jetbot #399: Newbee jotbot dad.

tjp1940, 2021-03-10:
I apologize if this is the wrong place to discuss my issue. Just starting and very little experience with forums. That being said, my jetbot works fine, however after running the basic_motion jupyter notebook I hosed up my Notebooks directory.  Is no longer exists in the  Desktop.  Can someone please explain how I can restore the Desktop/Notebooks with the demo notebooks back into the Desktop.  I suspect this is easy to do but I haven't figured it out yet. I have spent several days reading various documentation and visiting forums.

jaybdub, 2021-03-20:
Hi tjp1940,

Thanks for reaching out!

The notebooks' source code is located here https://github.com/NVIDIA-AI-IOT/jetbot/tree/master/notebooks

You could

1.  Clone this github repo:  Open a terminal and type ``git clone https://github.com/NVIDIA-AI-IOT/jetbot``
2. Copy the contents of the downloaded notebooks into your repo ``cp -r jetbot/notebooks <path to place>``.  

Please let me know if this helps or you run into any issues.

Best,
John

tjp1940, 2021-03-20:
Thank you very much for the suggestion.   I sorry I do not know how to open
a terminal with the nano.  When powered up all I get is a green light and
nothing else.  Using Putty on my PC I connected a USB serial cable to the
nano.  After I get the read out (see below) in my terminal screen, the
cursor just sits there.  I can't get any further. The keyboard does not
accept input. Is there a special sequence of keys to enter?

just

tjp1940, 2021-03-20:
sorry, here's my terminal message:
[0000.163] [L4T TegraBoot] (version 00.00.2018.01-l4t-e82258de)
[0000.168] Processing in cold boot mode Bootloader 2
[0000.173] A02 Bootrom Patch rev = 1023
[0000.177] Power-up reason: pmc por
[0000.180] No Battery Present
[0000.182] pmic max77620 reset reason
[0000.186] pmic max77620 NVERC : 0x40
[0000.189] RamCode = 0
[0000.191] Platform has DDR4 type RAM
[0000.195] max77620 disabling SD1 Remote Sense
[0000.199] Setting DDR voltage to 1125mv
[0000.203] Serial Number of Pmic Max77663: 0x2434f6
[0000.211] Entering ramdump check
[0000.214] Get RamDumpCarveOut = 0x0
[0000.217] RamDumpCarveOut=0x0,  RamDumperFlag=0xe59ff3f8
[0000.222] Last reboot was clean, booting normally!
[0000.227] Sdram initialization is successful
[0000.231] SecureOs Carveout Base=0x00000000ff800000 Size=0x00800000
[0000.237] Lp0 Carveout Base=0x00000000ff780000 Size=0x00001000
[0000.242] BpmpFw Carveout Base=0x00000000ff700000 Size=0x00080000
[0000.248] GSC1 Carveout Base=0x00000000ff600000 Size=0x00100000
[0000.254] GSC2 Carveout Base=0x00000000ff500000 Size=0x00100000
[0000.260] GSC4 Carveout Base=0x00000000ff400000 Size=0x00100000
[0000.266] GSC5 Carveout Base=0x00000000ff300000 Size=0x00100000
[0000.272] GSC3 Carveout Base=0x000000017f300000 Size=0x00d00000
[0000.288] RamDump Carveout Base=0x00000000ff280000 Size=0x00080000
[0000.294] Platform-DebugCarveout: 0
[0000.297] Nck Carveout Base=0x00000000ff080000 Size=0x00200000
[0000.303] Non secure mode, and RB not enabled.
[0000.307] BoardID = 3448, SKU = 0x0
[0000.311] QSPI-ONLY: SkipQspiOnlyFlag = 0
[0000.314] Nano-SD: checking PT table on QSPI ...
[0000.319] Read PT from (2:0)
[0000.334] Using BFS PT to query partitions
[0000.338] PT: Partition TBC NOT found !
[0000.342] Warning: Find Partition via PT Failed
[0000.346] BoardID = 3448, SKU = 0x0
[0000.349] QSPI-ONLY: SkipQspiOnlyFlag = 0
[0000.353] Nano-SD: checking PT table on QSPI ...
[0000.358] PT: Partition TBC NOT found !
[0000.361] Warning: Find Partition via PT Failed
[0000.366] Error is 1

tjp1940, 2021-03-20:
Me again.  Some clarification is needed here.  I had forgot I submitted an
item earlier #399,  Since then I submitted item #405.  Not mentioned
earlier was that the nano does not respond with any boot up action.

closed 2022-07-15 by jaybdub