NVIDIA-AI-IOT/jetbot #406: different images are not found / building of containers not possible

Jonathan-hdbw, 2021-03-23:
When following the steps from the [documentation](http://jetbot.org/v0.4.3/software_setup/docker.html), the following errors occur: 

```
jonathan@jonathan-jetson:~/jetbot/docker$ ./enable.sh $HOME
JETBOT_BASE_IMAGE not found for 32.5.1.  Please manually set the JETBOT_BASE_IMAGE environment variable. (ie: export JETBOT_BASE_IMAGE=...)
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Unable to find image 'jetbot/jetbot:display-0.4.3-32.5.1' locally
docker: Error response from daemon: manifest for jetbot/jetbot:display-0.4.3-32.5.1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.
Unable to find image 'jetbot/jetbot:jupyter-0.4.3-32.5.1' locally
docker: Error response from daemon: manifest for jetbot/jetbot:jupyter-0.4.3-32.5.1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.
```

Also when trying to run docker/build.sh, only the following error appears: 

```
jonathan@jonathan-jetson:~/jetbot/docker$ ./build.sh $HOME
invalid argument "/jetbot:base--" for "-t, --tag" flag: invalid reference format
See 'docker build --help'.
./build.sh: line 2: cd: models: No such file or directory
./build.sh: line 3: cd: display: No such file or directory
./build.sh: line 4: cd: jupyter: No such file or directory
./build.sh: line 5: cd: camera: No such file or directory
```

How to fix the error?

Hoehlenbaer, 2021-04-08:
Hi together,
For me it looks like the files on Github as well as the files in the Jetbot folder on the robot have the wrong version. All docker files are still referencing L4T 32.4.3, while Jetbot version 0.43 has been built using L4T 32.5.1. In addition the configure.sh script can't distinguish between 32.5.1 and 32.5.0, so it will never find a matching base image (as it looks like there's no pytorch-32.5.1 image).
Can you provide the right files on Github? I don't want to modify the build script by myself, because I don't know which versions the python libraries need to be.
In addition it might be helpful to build the camera and model images on the SD image as well. In this case one could easily run the already existing images without the need to go to the whole build process.

SatyaSudheer, 2021-04-14:
JETBOT_BASE_IMAGE not found for 32.5.1.  Please manually set the JETBOT_BASE_IMAGE environment variable. (ie: export JETBOT_BASE_IMAGE=...)
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Unable to find image 'jetbot/jetbot:display-0.4.3-32.5.1' locally
docker: Error response from daemon: manifest for jetbot/jetbot:display-0.4.3-32.5.1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.
Unable to find image 'jetbot/jetbot:jupyter-0.4.3-32.5.1' locally
docker: Error response from daemon: manifest for jetbot/jetbot:jupyter-0.4.3-32.5.1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.

Can someone help me, what base image I need to use.

Taitan2010, 2021-05-07:
I am facing the same issue with you. The error message commented by Mr.SatyaSudheer is displayed. 

**Considering the following insight, does the installation of the following 32.4.3 drivers effective?**
https://developer.nvidia.com/embedded/linux-tegra-r32.4.3

> Hi together,
> For me it looks like the files on Github as well as the files in the Jetbot folder on the robot have the wrong version. All docker files are still referencing L4T 32.4.3, while Jetbot [...]

Taitan2010, 2021-05-07:
**Or can we solve the problems if we replace the following programs to the scripts on the following URL?**

docker/base/build.sh
docker/configure.sh

https://github.com/NVIDIA-AI-IOT/jetbot/commit/ea7c7cdf0df8ba05ec56d3bf5c4d368cd43c3012

bkircher67, 2021-05-10:
Hello, 

after manipulating the configure.sh file I was able to experiment with my jetbot.

```
export L4T_VERSION="32.5.0"

if [[ $L4T_VERSION = "32.4.3" ]]
then
        JETBOT_BASE_IMAGE=nvcr.io/nvidia/l4t-pytorch:r32.4.3-pth1.6-py3
elif [[ "$L4T_VERSION" == "32.4.4" ]]
then
        JETBOT_BASE_IMAGE=nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3
elif [[ "$L4T_VERSION" == "32.5.0" ]]
then
        JETBOT_BASE_IMAGE=nvcr.io/nvidia/l4t-pytorch:r32.5.0-pth1.6-py3
else
        echo "JETBOT_BASE_IMAGE not found for ${L4T_VERSION}.  Please manually set the JETBOT_BASE_IMAGE environment variable. (ie: export JETBOT_BASE_IMAGE=...)"
fi
```

My experience was that L4T 32.5.1 didn't work at all - so I faked it to 32.5.0 

KR Bernd

Taitan2010, 2021-05-12:
Dear Bernd
Thank you for your comment. If we write the above program somewhere, does it mean that Jetson Nano of 2GB Type can also be used? If so, where should it be written?

Taitan2010, 2021-05-14:
> Hello,
>
> after manipulating the configure.sh file I was able to experiment with my jetbot.
>
> ```
> export L4T_VERSION="32.5.0"
>
> if [[ $L4T_VERSION = "32.4.3" ]]
> then [...]

Dear Bernd
Thank you for your comment. If we write the above program somewhere, does it mean that Jetson Nano of 2GB Type can also be used? If so, where should it be written?

bkircher67, 2021-05-14:
Hello Taitan2010,

sorry, as I don't have the 2GB version I cannot guaranty that my workaround works with this hardware (I'm using the original 4GB version).

But according to nvidia they say that:

* NVIDIA L4T 32.5.1 supports all Jetson modules: Jetson AGX Xavier series, Jetson Xavier NX, Jetson TX2 series, Jetson TX1, and Jetson Nano. All Jetson developer kits are also supported.

and

* L4T 32.5.1 is identical to L4T 32.5 <https://developer.nvidia.com/embedded/linux-tegra-r325> except for the following new features:

Developer preview support for Jetson TX2 NX module
Developer preview of pre-built Real-Time Kernel package <https://developer.nvidia.com/jetson_linux_driver_package_release_notes_r32.5.1_ga.pdf> for Jetson AGX Xavier

 
So basically it should work.

KR Bernd

nikhilnnk, 2021-06-21:
Use SD card image for docker given in the [link ](https://jetbot.org/master/software_setup/sd_card.html).
Note that normal software installation of jetson nano should be done before going to the docker image file booting.
Hope it helps :)

santaimpersonator, 2021-06-29:
The workaround from @bkircher67 works with the latest 2GB Jetson Nano pre-built image: [jetbot-043_nano-2gb-jp45.zip](https://drive.google.com/file/d/1tsuSY3iZrfiKu4ww-RX-eCPcwuT2DPwJ/view?usp=sharing) on [jetbot.org](http://jetbot.org/master/software_setup/sd_card.html).

In the bash history of the pre-built image `nvcr.io/ea-linux4tegra/l4t-pytorch:r32.5.0-pth1.7-py3` is used, but `nvcr.io/nvidia/l4t-pytorch:r32.5.0-pth1.6-py3` is compatible between both boards.

santaimpersonator, 2021-06-29:
@Srafington You will need to boot the board with the blank/plain image first. The initial setup configuration will reflash the bootloader, which will allow the board to use the new pre-built image.

*See the call out at the bottom of **Step 1** here: http://jetbot.org/master/software_setup/sd_card.html*

closed 2022-07-18 by jaybdub

vnguyen13, 2023-06-15:
> Hello,
>
> after manipulating the configure.sh file I was able to experiment with my jetbot.
>
> ```
> export L4T_VERSION="32.5.0"
>
> if [[ $L4T_VERSION = "32.4.3" ]]
> then [...]

This worked for me. Thank you!

rockbabyg, 2023-12-21:
> Hello,
>
> after manipulating the configure.sh file I was able to experiment with my jetbot.
>
> ```
> export L4T_VERSION="32.5.0"
>
> if [[ $L4T_VERSION = "32.4.3" ]]
> then [...]

It's working, Thank you